RasPlex/RasPlex #481: Start of music tracks trimmed

Stewie-Griffin, 2016-04-06:
Whenever I'm listening to music, the beginning of the next track is constantly being trimmed. This is most noticeable with continuously mixed music, the playback is gapless but I'm missing roughly the first 0.5sec of the next track.

I've linked the first 4 tracks of a recently purchased album, there is a noticeable jump when one track ends & the next begins but if I playback on a different device the playback is smooth.
﻿
https://www.dropbox.com/sh/ifod26gjpqfm1s7/AAB5bS-v2sQVCE2wrxA4gEFza?dl=0

I'm running Rasplex 1.0.3 on a Raspberry Pi2 with a Hifiberry DAC+ as a headless music player, I control playback with either the iPhone or the iPad app 4.0.15. Plex Media Server 9.16.3

This happens if I'm connected via Ethernet or Wifi, I have a class10 microsd card

Stewie-Griffin, 2016-04-08:
Some more info for you

RaspberryPi2 Serial: 00000000afc14767
Crash ID: None
Raspberry: Pi2 Model B 1GB
SD Card: SanDisk 8GB MicroSD Class 10
Connection: Ethernet or Wireless
Version: RasPlex 1.0.3
How did you install rasplex? Mac OSX GetRasplex app
Power Supply 5V 2A

NedtheNerd, 2016-04-08:
Im still trying to replicate this issue, is your RPi/DAC connected to an amp or AVR? Have you tried controlling playback from PlexWeb, if so do you get the same issue?

Stewie-Griffin, 2016-04-08:
Hi Ned,

The RPi/DAC+ is connected directly to an Arcam Alpha 9 via phono leads. I have just tried playback via the PlexWeb and I'm still getting the same results - the initial downbeat of every track is trimmed. One thing I did notice when trying to use the web app, the playhead would reach the end of the track and indicate the start of the next track a fraction of a second before the jump/trim occurs.

I'm trying to playback content from two different servers, one local hard wired into the network and another remote server. It doesn't make a difference which server I'm trying, they both cause the Rasplex to skip the downbeat.
 
The only items connected to my RPi2 other than my amp are a FLIRC dongle and a WIFI dongle, I've attached a link to both below.

http://www.amazon.co.uk/TP-LINK-TL-WN823N-Mbps-Wireless-Adapter/dp/B0088TKTY2/ref=sr_1_1?ie=UTF8&qid=1460122445&sr=8-1&keywords=300mbps+wireless+adapter

http://www.amazon.co.uk/FLIRC-Dongle-Media-Centre-Raspberry/dp/B00BB0ETW8/ref=sr_1_1?ie=UTF8&qid=1460122465&sr=8-1&keywords=flirc

If it would help I can link some more music with more of a steady beat that has a much more noticeable jump for you to try?

NedtheNerd, 2016-04-08:
This could be caused by the audio device not being kept alive on change of tracks, I have a Marantz and a Sony amp and neither demonstrate this issue, but then not all amps are the same.  In the next release there is a setting in Audio Preferences to set a time to keep the audio device alive and can be set in numbers of seconds.  Not saying that this is the issue but it may be.

Stewie-Griffin, 2016-04-09:
Ok, I'll wait for the next release and see if there's any improvement.

Thanks

PS. Don't know if this would make any difference, I do have my pi2 overclocked with the following settings

arm_freq=1000
  core_freq=500
  sdram_freq=483
  over_voltage=2

NedtheNerd, 2016-04-10:
Shouldn't really need any overclocking and this shouldn't affect what you are experiencing, no harm in removing and restarting.

Stewie-Griffin, 2016-04-15:
Since my last post I have tried a full fresh reinstall of 1.0.3 taking the microsd card to my computer and using the GetRasplex OSX app to wipe the card and start again. Unfortunately this has not fixed the problem.

I have linked to a folder with some mp3 audio examples for you to hear what I'm hearing from Rasplex and what the file should sound like with no playback error. "what rasplex plays" was recorded directly from the HifiBerry outs connected to my audio interface then converted to mp3 to save size, the skipping error occurs at around 41 seconds. The "what should play" file was recorded directly from Swinsian using a loopback in my audio interface then converted to mp3 to save size.

https://www.dropbox.com/sh/cbdfwu8fq60q5pr/AAB1RWJMWgVlJjipBAEHicTta?dl=0

I have also included the two audio files I used for the recording so you can double check that you are not hearing the skip on your system. 

I know you have already stated that you do not have the problem but I thought there might be a possibility that if you heard my two examples then tried with the same tracks maybe you would also have the same issue.

NedtheNerd, 2016-04-15:
Thanks, will have a look.

NedtheNerd, 2016-04-29:
Confirmed fixed in next release 1.6.0

NedtheNerd, 2016-05-07:
Fixed using, option to 'Keep audio device alive' in Preferences->System->Audio output in next release.

Issue relates to Arcam amps and DACs, does not only affect Rasplex but also other software that outputs audio via a DAC to Arcam amps, may affect other amps but confirmed not an issue with Marantz or Sony.

NedtheNerd, 2016-06-06:
Can you confirm fixed in just released 1.6.0, thanks.

Stewie-Griffin, 2016-06-08:
Done.

This issue can be closed now, thanks

closed 2016-06-08 by Stewie-Griffin