RasPlex/RasPlex #482: [Enhancement] Build libass with HarfBuzz enabled

SafaAlfulaij, 2016-04-08:
Arabic shaping break if HarfBuzz isn't used (issues differs from a font to font).
Issue describing the problem (images included): https://github.com/libass/libass/issues/134 (The problem is with Aegisub for Windows as well)
An issue for SuperTuxKart regarding supporting Fribidi (Done in Rasplex) and Harfbuzz: https://github.com/supertuxkart/stk-code/issues/2320

Example of using Fribidi without HarfBuzz:
[image]
Same example but the HarfBuzz used (Notice the diacritics and some other characters (check: https://en.wikipedia.org/wiki/Arabic_alphabet#Ligatures):
[image]

Please let me know if you want a detailed description

SafaAlfulaij, 2016-04-15:
I see the label "Done" there, is it done and you are testing it or what? :)
Can you grab me the patch or the RaspberryPi's ISO to check?

NedtheNerd, 2016-04-15:
You will have to wait for the Pre-release version.

SafaAlfulaij, 2016-04-17:
I just like to add something, when the subtitle (SSA) is embedded into the MKV file (in my case), the text renders correctly and without any problems, but when it's side by side to the video it does not.

NedtheNerd, 2016-06-06:
Can you try with latest 1..6.0 release and advise.

SafaAlfulaij, 2016-06-07:
Unfortunately, still not rendering the text correctly  :(
Can you lead me to the changes you made? I may have an idea...
Thanks for your work!

NedtheNerd, 2016-06-17:
Release 1.6.1 is out, can you check, hopefully this issue is nice fixed for you.

SafaAlfulaij, 2016-06-17:
Yeah! That fixed it! Huge thanks!

closed 2016-06-17 by SafaAlfulaij