RasPlex/RasPlex #484: [BUG] Play bar during Movies/TV Shows does not hide itself after inactivity (1.0.3)

sashahilton00, 2016-04-14:
Serial: 000000000f62db87, 512MB Pi 2, SDHC class 10, Ethernet, Rasplex Ver. 1.0.3, OTA update from 1.0.2.

The problem I am seeing at the moment is minor but moderately irritating, and it is that when pausing a movie or TV show, the playback bar appears as normal, though once un-paused, the playback bar never re-hides itself, meaning that one has to stop the movie, go back to the pre-play screen, then resume the said movie/TV Show in order to continue playback without the playback bar at the bottom of the screen.

Image of problem attached.

[image]

P.S. It's not me being impatient, I have continued playback for 5 minutes to check :)

NedtheNerd, 2016-04-26:
This issue is specific to your installation and is not a bug as this is not normal and cannot be replicated.

Suggest reflashing your SD card as there may be some corruption causing this.

Changing title and closing issue.

closed 2016-04-26 by NedtheNerd

sashahilton00, 2016-04-26:
Apologies. Tried the reflash at first and it persisted, though reflashing again resolved it. Probably a dodgy SD card