RasPlex/RasPlex #485: RP-1.0.3 Home Movie type libraries display posters in landscape mode.

trumpy81, 2016-04-15:
When entering a Home Movie type library all posters are displayed in a landscape aspect mode.

My libraries have posters for all video files and they only display the top half of the poster which makes my libraries look awful and it also makes it difficult to identify the file you are looking for.

When viewing Home Movie type libraries in PMS, posters are correctly displayed in portrait mode.

See this thread: https://forums.plex.tv/discussion/comment/1167527#Comment_1167527

NedtheNerd, 2016-06-06:
Should be fixed in latest release 1.6.0, please advise.

trumpy81, 2016-06-07:
No, not fixed in 1.6.0, I still get landscape orientation instead of Portrait, maybe this is a PMS issue?

NedtheNerd, 2016-06-08:
Is this with default skin or one of the others?

trumpy81, 2016-06-09:
It's with the Default Skin.

NedtheNerd, 2016-06-09:
I think the skin change was made to the Aeon Nox skin, but the change didn't get into the default skin, can you have a look? Thanks

NedtheNerd, 2016-06-09:
Can you also try this Plex Black Edition skin https://goo.gl/iqZ8OE

trumpy81, 2016-06-09:
I installed Aeon Nox (yuk!) and after it downloaded and installed itself, I entered one of my Home Movie libraries, but it still displays in landscape mode instead of portrait mode, as shown in the screenshot.

The same applies to the Plex Black Edition Skin and the OpenPHT Skin, they also display my Home Movies library in landscape mode.

I should also mention that the overscan (screen calibration) settings are not being saved.

Aeon Nox
[image]

Plex Black Edition
[image]

OpenPHT
[image]

magnumdoomguy, 2016-06-10:
Displaying Home Video posters as portrait instead of landscape would break it for the intended use of home videos.  When adding actual home videos to PMS it creates landscape thumbs for them.  

iOS also displays home videos in landscape, and would look similar to your screenshots above.

Poking around in themoviedb.org I could find most of the titles from your screenshots above.  I'm baffled as to why you'd have these titles in a home video library.

I see no reason to break home videos in OpenPHT for their intended use, to make it look better for your unusual use.

closed 2016-06-10 by magnumdoomguy

trumpy81, 2016-06-10:
Sorry to say, but Home Video libraries have always displayed in portrait mode (up until RasPlex 1.0.3), that is why I created all of my posters in portrait mode long ago, even before I owned an RPi2.

Displaying in landscape mode is a recent development which now breaks all of my posters, a lot of which were created by me. I don't give a rats what iOS does, that is not even relevant to this issue.

And now you are trying to tell me how to organise my libraries, good luck with that.

And FYI, this is not the only Home Movies library I have.

In Plex/Web, all Home Movie libraries still display in portrait mode as can be seen in the screen shot attached here:

[image]

mastyman, 2016-06-10:
I disagree with above, even tho Plexweb shows it as a portrait, its still natively a landscape image & Plexweb is making a mess of it really,  Plexweb should be changing it to landscape like in other plex apps and openpht etc, as thats what makes sense for Home Video library types.. 

I certainly dont have to have to MANUALLY create new posters for thousands of home videos because of this guy

trumpy81, 2016-06-11:
And that's exactly how I feel, why should I have to recreate all my posters just to satisfy an issue with RasPlex/OpenPHT?

Like I said, RasPlex used to display my posters correctly in portrait mode until 1.0.3, now I am being forced to change them, why?

Not all actual Home Movies need to be presented in landscape mode, so why should the posters be presented that way?

As for Plex/Web, again, that is the way Plex/Web has always displayed posters and that is why my posters are in portrait mode to begin with.

I don't see any reason why an option could not be added to settings/skins to allow both orientations for Home Movie libraries. That way, everyone would be happy and no one would need to change posters.

magnumdoomguy, 2016-06-11:
The proposed option to allow both is far more acceptable than making home videos look worse for the majority of users so that they look better for a minority of users.  I'd be willing to consider coding this into the default skin.

We're just volunteers putting our time into a rather cool project.  As a "Plex Ninja" I doubt that you write as brusquely to the Plex Employees when requesting a feature that would require a few days of coding.

trumpy81, 2016-06-11:
Why wouldn't I, do you think they cut me any slack because I'm also a volunteer?

If someone wants to butt heads, then I will oblige, most of the time ... lol

I'm a grouchy old codger and sometimes that is reflected in my ramblings, even though I don't always intend it to be that way.

Apologies if I have offended anyone in any way.

I'm not sure exactly what changed in the code but something obviously did and that effected the way posters are displayed in Home Movies libraries. I would be perfectly happy to have the option, even if the default was landscape. I certainly don't mind making a settings change. It can be frustrating when you don't have such options though.

And since you were wondering, the majority of my videos in that documentaries folder are not recognised by either TV or Movie databases (Humphrey Jennings videos and some others) or at least they weren't at the time I added them, that's why they were relegated to a Home Movies library.

Since then, many of the newer documentaries were added to the same library simply to make them easier to access. It can be difficult to locate a file amongst 5600+ other Movie files. :)

NedtheNerd, 2016-06-11:
I think we know have a way forward, can we keep posts to the point, sorry, but I have edited recent posts to keep it constructive.

NedtheNerd, 2016-06-18:
@trumpy81 

Default skin now has preferences option to show posters as portrait in 1.6.1, other skins will get the change soon.

trumpy81, 2016-06-18:
Thanks Ned and thanks to all who made that happen, it is very much appreciated and it is working great!!. :)