RasPlex/RasPlex #486: Rasplex subtitle problem

getter3, 2016-04-22:
I encounter some problem with the subtitle and the movie information

I am using raspberry pi 2 model B+ 
I must say that I am mainly using Chinese (traditional or simplified) subtitle and movie information
Things running smooth on start up, subtitle and movie info shown up normally

But after running some time (like 20-30mins) , the subtitle and movie information screwed up.
It turned into a "white square block" randomly~~with sound burst and jittering
And after sometimes, the "white square block" persist

I have also tried other subtitle , even the english subtitle, also with the above problem
I have also tried the latest **1.0.3 pre-release**, same problem here

So is there any way or settings to prevent this problem?﻿

More information is that I have tried to use Kodi on my raspberry pi with plexbmc plugin, no problem so far

NedtheNerd, 2016-06-06:
Can you try this with late set release 1.6.0, thanks.

getter3, 2016-06-13:
I have tested with 1.6.0

The subtitle is not turned into white block and no sound jitter

But other problem occurs
1. The video lag/slow a bit whenever each lines of subtitle comes out. The larger the font size,the more significant the lag (font size above 20+ with more apparent lag)

2. When scrolling the playlist or screen fast , the whole screen wording turns into white block
It can only recover by restart the application

NedtheNerd, 2016-06-17:
Can you check in newly released 1.6.1 and advise.

NedtheNerd, 2016-06-27:
No further responses, closing issue

closed 2016-06-27 by NedtheNerd