RasPlex/RasPlex #487: RasPlex 1.0.3 system reboot when trying to view video

mgollo, 2016-04-24:
Basic information:
RPi serial number: not available to me at this point
Crash id: 826
Size of RPi: reproduced on RPi2 with 256 and RPi3 with 512
Type of SD card: Class 10 micro SD on RPi3 and Class10 SDHC on RPi2
Type of network connection: Onboard WiFi on RPi3, WiFi dongle on RPi2. WiFi AP was less than 2m away, so the signal was fine.
Version of RasPlex: 1.0.3
Installed via installer on Mac OS X
Type of Plex system: Synology NAS

Problem description: When selecting any movie, the system reboots when I try to start viewing it. Strangely, the video codec is reported as unknown. The tested movies play fine using an Apple TV as a client (even with a weaker WiFi signal). This is reproducible with any movie on my Plex that I have tested so far. I also tried it with a different RPi (see above), the behaviour is identical.  Attaching the debug log (from the RPi 3) does not work, I pasted it here: http://pastebin.com/MhMkAUWt.

g1t-dlanor, 2016-04-24:
The information on your RPi RAM sizes seems incorrect.
I think you may be mistaken about the RPi models you have.

RPi1A has 256 MB  RAM
RPi1B early release also has 256 MB RAM (very few)
RP1B normal release has 512 MB RAM
RPi1B+ has 512 MB RAM
RPi2 and RPi3 have 1 GB RAM

So if the RAM sizes you stated are correct, then the one with 256 MB must be an RPi1 of the early models which are not supported by RasPlex. And the one with 512 MB must be a later model of RPi1, which is supported by RasPlex, but which will not work with the RPi2 SD card images.

Further proof that one of your units is an RPi1 is given by the fact that it  uses a normal SDHC card, since both RPi2 and RPi3 require micro SD cards. But that is also the case for the RPi1B+, which has the same physical appearance as the RPi2, but still has just 512 MB RAM. On the other hand, the RPi3 is the only model which has onboard WiFi, so if you have that there is no doubt about its being an RPi3, and you're just  mistaken about its RAM size.

Please double-check what RPi models you have and make sure to use the corresponding RasPlex images for each. (Though an early RPi1 with just 256 MB is doomed to fail anyway... :( )

mgollo, 2016-04-24:
Yes, you are right. I was a bit confused by the different models I have lying around here. Indeed, it was a RPi1B with 512MB RAM and an RPi3 with 1024MB, of course. My bad, sorry for the confusion. However, the RPi1B worked with v1.0.1 with the same videos, albeit very slowly (which is why I now switched to the RPi3).

g1t-dlanor, 2016-04-24:
OK then, with that cleared up we can focus on the real issue, of your RPi units crashing on playback.
I'm quite surprised at that result, as I and many others have used that RasPlex version with good results on both RPi1 and RPi2, and others have also reported it working well on the RPi3.
So it should be something specific to your setup that triggers this crashing.

The two most common causes for crashing are SD card failures (mis-flashed or otherwise corrupted) and insufficient supply power (underpowered PSU or  low-quality micro-USB cable with too high resistance). The RPi3 is especially demanding when it comes to PSU current, but both devices will draw the most current during playback, as both CPU and GPU may be working hard then. So that could be one explanation why video launch triggers the crash.

In your place I'd reflash the SD cards from scratch, preferably using other cards than the ones you used during the crashes, and also double-check that the PSU adaptors deliver sufficient current.

mgollo, 2016-04-25:
Thank you for your feedback. I'll try to do the reflash and testing ASAP. Regarding the PSU: I used the one that I ordered with the RPi1 (it's an official one), which delivers 2A. I read that for the RPi3, 2.5A are recommended, but since I have the same behaviour with the RPi1 that consumes less (and I even tried to replace the PSU with an iPad PSU with 2.0A as well), I don't really think that this should be the problem.

Thanks again and I'll get back to you as soon as I have new information.

Reedler, 2016-04-26:
I'm having the same issue with video playback and especially airplay. I'm running a RPi3. I actually tried this both of my RPi3 and I've tried a Samsung 32GB micro SD class 10 and a Sandisk 32GB class 10 card. I'm running 1.0.3. Was watching Silicon Valley a .mkv file and it rebooted. After the show I tried to airplay some music and it keeps rebooting. What's the default ssh username and pw so I can view the log files? I tried the default root and rasplex but that didn't work.

Reedler, 2016-04-26:
FYI I didn't use the installer I downloaded the image file and used dd to copy the image over.

Reedler, 2016-04-26:
Also a little more background. When I play music directly from rasplex, every 10 seconds the music stops and after a few minutes my RPi3 reboots. I'm on a full gigabit network for the devices that have ethernet cards. I have the Plex app on an Amazon Fire Stick, a Roku Stick and a PS3. I'm not having Plex app issues video or music with any of these devices. I also have airplay installed on my other RPi3 on top of Jessie and I have no issues with airplay.

g1t-dlanor, 2016-04-26:
@Reedler:
Since you describe your issue as tied to to airplay  and it even affects pure music playback I strongly doubt that your issue can be the same as that of @mgollo. So you should open your own issue for your problem or find another existing one that matches your problem better. Otherwise you will just clutter up this issue that he opened, thereby making it harder for him to get accurate help. Here I will just address a few details of your problem, but I don't want this to grow into an ongoing discussion here. Opening a new separate issue is the correct way to deal with it.

For now I just have a few comments on your problem as described so far:
Since you have no such problem on other Plex clients we can probably exclude the PMS server from responsibility for the issue. In any case a pure server problem should only cause playback or library access failures, never crashing of the client device. Thus the cause of the problem must be located at the crashing RPi3, though not necessarily 'in' it. As always with RPi units two main suspects are possible corruption of the SD card and insufficient or unstable power supply (not necessarily caused by the PSU itself, as USB cables with too high resistance can have the same effect). Unfortunately I can't speak from personal experience of RasPlex functionality on the RPi3, since I don't have that version yet (awaiting delivery of a pi-topCEED computer which includes RPi3), but other members of the RasPlex team have reported RasPlex as working fine on the RPi3, so it should be possible to achieve the same on yours as well. As for airplay from IOS devices, that has always been 'iffy' due to variations in Apple's airplay protocols for different IOS versions making it hard to keep up. But again, this kind of failure should not cause device crashing, so that is a definite issue.

About  your install method:
"dd" may be powerful, but that in itself is a risk, as small mistakes can cause powerful errors.
I recommend that you try a new installation, using either the standard RasPlex installer or some other simple GUI tool, such as "Win32 Disk Imager" under Windows. Those are what I use myself.

Finally in reply to your query on SSH:
The default username and password for RasPlex SSH are "root" and "rasplex"
(without quote chars and all lowercase)

These have been the defaults for a long time, and still remain unchanged.
Failure for these to work may be due to some firewall in your computer, or possibly some 'certificate censorship'. For example: Each time I use the PuTTy program to connect to a newly installed  RasPlex system, PuTTy gives a warning about faulty/changed SSH server certificate, which I must override to make PuTTy accept the connection as valid (remembered until I reinstall RasPlex again).
Some other SSH clients may check certificate in similar ways, but not offer the user an override...

I hope some of the above can help you, but if not then please continue this discussion in another issue thread.

NedtheNerd, 2016-04-26:
@Reedler 
For AirPlay please see issue @456

NedtheNerd, 2016-04-26:
Failure to play video or music is not a common issue or bug, as @g1t-dlanor has stated this is more likely to be connected to your installation, suggest the following:

Reflash a new SD card with latest release, do not copy any settings files from your current installation and reconfigure System Settings and Preferences, if you have SD card corruption issues then these can permanently damage the SD card, especially if you pull the power cord without executing Shutdown first.

Check you power supply, minimum 2A for RPi0, RPi1 and RPi2, minimum 2.5A for RPi3, if using a USB power supply with a USB to micro USB cable, check the cable, thin cheap or cables longer than 1m can cause significant power drops causing rebooting.

Check you firewall settings on your router.

Closing issue.

Reedler, 2016-04-26:
@g1t-dlanor and NedtheNerd, thanks for the quick responses. I just rec'd these two pi's within the last few weeks and both are using the default power supplies. I also stated in my previous post that this is happening with video playback as well, TV shows and Movies. Last night I re-downloaded the image and used the OSX installer and flashed another brand new micro SD card(class 10), kept all the default settings. I'm still having the same issue with video playback. I have no firewall running on this mac and I'm not using any ssh "cert censorship". I'm also not having any rebooting issues when I'm running Jessie, OSMC, and Openelec. This is why I posted here as I'm having basically the same issue as mgollo in the original post. Please don't close as I just wanted to give you some more background and info on this. I was also tailing my PMS logs as I was connecting to see if I could see any errors. Thanks!

g1t-dlanor, 2016-04-26:
@NedtheNerd:
I think you mistakenly closed the issue, forgetting that we still have to deal with the issue of @mgollo, whose response to the latest advice we're still waiting for. That's why I reopened the issue now.

@Reedler:
Your issue is clearly NOT identical to that of @mgollo, since you reported additional problems that he did not (such as reboots for music playback and airplay issues). That's why I asked you to not clutter up his thread, as that may prevent him from getting accurate help. Make your own issue thread instead, unless you can find an existing one that better matches your entire issue. Then we can deal better with that issue as well.

As for the SSH problem, something is seriously wrong if you can't log in to the SSH server using the default 'root' + 'rasplex' username and password strings. These are the ones defined for the current RasPlex versions. And about the 'SSH cert censorship', I naturally did NOT mean that you used any such intentionally, but rather that the software you use has some such feature as a built-in default, just like PuTTY has (though it fortunately allows user override). And if your RPi is connected to the Mac through some smart LAN router/switch, that too might interfere with the SSH protocol (commonly blocked as potential hacking tool). I'm not saying anything definite about what it is that interferes with your SSH traffic, only that something clearly is doing so.

Reedler, 2016-04-26:
Ok, so after reflashing last night using the installer I am able to ssh into the rasplex so something was definitely wrong with the image. I was too tired to check that last night. I'm still having video playback reboots.

NedtheNerd, 2016-04-27:
Logs!!!!!!!

mgollo, 2016-04-29:
After re-flashing the RPi, nothing had changed, but the problem was not caused by RasPlex. An upgrade of the Plex server on the NAS caused the problem to disappear. So it was not RasPlex's fault, but a bug on the server side. Sorry for causing a fuzz.

p.s. Additional information for possible future reports: Some Synology NAS systems do not have the necessary computing power to transcode videos in realtime. This can be a cause for slow or stuttering video playback as I found out during my research – especially older devices are affected because they cannot transcode in hardware and therefore suffer from high CPU load, making them really slow.

closed 2016-04-29 by mgollo