RasPlex/RasPlex #488: RasPles 1.03- 640x480+overscan on HDMI after power cycle

therealjohnsnow, 2016-05-04:
This issue is present on both my RPi1 and RPi2 devices running the latest Rasplex 1.03.

I have not encountered this in the past 2 years and it is relatively recent with no change in equipment or gear.

When I turn off the main (the TV, AV receiver and RPi) and turn them back on, the rasplex screen comes up, but in low res 640x480 and overscanned.

At this point I will restart the Rasplex alone via the menu option and after this everything will be normal (1920x1080p 60Hz) again without me having to touch the settings or preference at all.

I have reproduced this many times on my setup. It only occurs the first time, from a 'cold boo i.e main switched on. After I restart the rasplex alone, the issue resolves.

Any idea?

Thanks.

Kwiboo, 2016-05-04:
If you only use your Raspberry Pi device on a single TV/AVR setup I suggest you follow the steps under _2.7 TV is not detected unless powered on first_ on http://kodi.wiki/view/Raspberry_Pi_FAQ#TV_is_not_detected_unless_powered_on_first to force your RPi to remember the EDID information from your TV/AVR.

NedtheNerd, 2016-06-06:
This is TV dependant, not an issue with Rasplex, as Kwiboo has stated please follow the link he posted for ways to resolve this.  Closing issue.

closed 2016-06-06 by NedtheNerd