RasPlex/RasPlex #489: Only main account in visible on boot up (Plex Home)

therealjohnsnow, 2016-05-04:
I have setup Plex Home and it works well enough on other devices. But seems like not working right here.

On first boot of rasplex, only the main account is visible and available for login. 
Only after that, am I able to switch user to other users.

Shouldn't other users be able to log in straight away on boot up? 

Thanks.

Kwiboo, 2016-05-04:
This sounds like a time and/or network issue, try with the "wait on network" option enabled in System Settings and possibly specifying a time-server. Are you using WiFi or Ethernet?

NedtheNerd, 2016-06-08:
Is this still an issue in release 1.6.0 with Wait for Network enabled?

therealjohnsnow, 2016-06-08:
will get back to you on this with some further testing