RasPlex/RasPlex #490: [Feature Request] NTP.org pool servers set by default

benjaminwolf, 2016-05-05:
An initial default setting of pool.ntp.org in the NTP server options would save most users a little setup time. It has the added benefit of preventing new Rasplex users who are less technical from having time/date related encryption issues.

Users who do not want to use an NTP server or who have a specific NTP server they wish to use could then change the setting. There are fewer users needing custom NTP servers, making the setup slightly more efficient for the majority without impacting the few who actually need custom settings.

Information about the NTP.org Server Pool project can be found here:

http://www.pool.ntp.org/en/

Kwiboo, 2016-05-29:
The System Settings addon should already be configured with fallback timeservers from pool.ntp.org. However there is an issue in the System Settings addon that very easily causes the fallback timeservers to be ignored.

After one of the `Network -> NTP Servers -> Timeserver` options has been opened and closed without entering any value the fallback timeservers would be ignored.
The only way to recover from this issue is to enter a working ntp-server or manually removing the faulty `Timeservers` line from `/storage/.cache/connman/settings`

https://github.com/RasPlex/service.openelec.settings/commit/2d573f1fbb775c5e9218f06445866e4f8e93ec52 fixes this issue and will be included in the 1.6.0 release

Kwiboo, 2016-06-11:
To fix broken configuration after an upgrade to 1.6:
Open one of `Network -> Time server #` option and save in System Settings.

closed 2016-06-11 by Kwiboo