RasPlex/RasPlex #492: Rasplex causing Plex Media Server to crash on plugin usage

Helza, 2016-05-06:
I've been having issues with the latest plex media server crashing when i use the TwitchMod plugin.
I posted a message on the plex forums about this and it turns out rasplex is creating invalid url's when using surten plugins. Ever since i upgrade to plex media server 0.9.16.6 its just been silently crapping out when pressing the back button on twitchmod, but it also happens in the plexinc bbc iplayer for example.

Apperently there is some error in creating the urls with an invalid playQueueVersion parameter.

`May 05, 2016 14:01:00:751 [6140] ERROR - Had trouble breaking playQueueVersion
May 05, 2016 14:01:00:751 [6140] ERROR - ERROR: Parsing request failed.
May 05, 2016 14:01:00:751 [6140] DEBUG - Request: [192.168.2.6:52398] GET /:/timeline?containerKey=%2fplayQueues%2f&key=%2fsystem%2fservices%2furl%2flookup%3furl%3d1https%253A%252F%252Fwww.twitch.tv%252Friotgames&playQueueVersion&ratingKey=1https%3a%2f%2fwww.twitch.tv%2friotgames&state=stopped&time=0&url=1https%3a%2f%2fwww.twitch.tv%2friotgames&X-Plex-Token=xxxxxxxxxxxxxxxxxxxx (7 live) TLS
`

g1t-dlanor, 2016-05-07:
I agree that the RasPlex client should not crash, even if/when using a  3rd-party Plex channel which is apparently under development and also having issues with other Plex clients and PMS itself (as noted in recent forum posts). So for cases of abnormal channel interactions that a client can't handle, the ideal behaviour would be to detect the problem and abort that channel connection rather than crashing.

However, the mere fact that PMS also crashes indicates some serious abnormality in that channel, as well as some dubious coding of PMS itself. If a channel (or a client) behaves abnormally PMS should never crash, but should note and log the abnormality and stop the ongoing connection attempt. In this context it should also be noted that you yourself have described this issue as specific to the latest PMS release 0.9.16.6 in your own forum posts. So if the problem did not exist with a previous version of PMS, then the blame can not be entirely placed on RasPlex, though its specific behaviour might trigger a bug in PMS.

Naturally a client like RasPlex should also ideally detect and stop abnormal behaviour rather than crashing when such occur. But in any server/client relationship the main responsibility for preventing crashing interactions rests with the server, since that is central to all communication with all clients. And that applies to all channels as well, as their code is installed as part of the local application data of the PMS server, not installed in any clients.

So my conclusion is that all three of the involved softwares need to be fixed !!!

The TwitchMod channel clearly does not behave like other channels (needs to be normalized).
The RasPlex client clearly can't handle channel problems correctly (should NEVER crash).
The PMS server clearly can't handle channel problems correctly (should NEVER crash).

And since older PMS versions didn't have this problem with this channel (according to your own posts), some recent change needs to be reverted.

NB: Fixing just one of these 3 softwares will probably not fix the entire problem.

NedtheNerd, 2016-05-07:
Can confirm this issue with other channels, BBC iPlayer channel (Plex supported) plays a video OK, but on stop it returns a Get Directory failed in Rasplex and restarts PMS.  Same channel, same video on Plex for iOS does not have the issue, will also check other clients. This appears to be the way Rasplex handles channels and not a PMS issue, works the same on both 1.0.3 and next release.

Helza, 2016-05-07:
Just a short clarification, it only crashes plex media server, not the client. 
The issue is in the "back" option if you use plugins and its not specific to the plugin itself. The plugin is just creating an ObjectContainer and adding DirectoryObject & VideoClipObject items, it does not setup the playQueueVersion in the url, thats handled by the client. Its those url's that are created invalid and make the server crash. Plex has already said they will fix the server crash in the next update, however that won't fix the underlying problem thats causing these url's to be created invalidly and will cause undeterminated results when they do release it.

saivajr, 2016-05-19:
Hello,
I am new to Rasplex, and Plex in general, but I have been using it for the past month and I have to say the support from you guys is great. My only issue at the moment would be the bug you have describe above. Is there a place to see if and when it's going to get fixed in future releases? I don't want to bother you with obvious stuff but I can't seem to find the correct place.

NedtheNerd, 2016-05-19:
@saivajr 

This is the place.

saivajr, 2016-05-19:
Haha, thanks for the quick reply. I'll try to stay updated from here then. Hope the issue gets solved soon. It can actually be bypassed if you press tab to go to the previous screen and select a different video but it's not really convenient. It looks like you are all over the place Ned helping everyone, keep up the nice work!

NedtheNerd, 2016-06-04:
Fixed in next release - Rasplex 1.6 and PMS 0.9.17.

saivajr, 2016-06-04:
That is great news. We'll wait for the release!

NedtheNerd, 2016-06-06:
1.6.0 is out, can you confirm this as fixed, thanks.

Helza, 2016-06-07:
Afraid it didnt fix it. Updated RasPlex to 1.6.0, restarted, watched a twitch channel for a bit and after stopping the channel and pressing exit it still crashed plex media server. The only difference was that the screen just went entirely black for a while instead of freezing in place before it went back to the main menu

sa2000a, 2016-06-07:
The server is not liking the `playQueueVersion` argument in the timeline request. Not sure what the syntax would be when the timeline is for a url.

Plex media Server 0.9.17.0 stops the crashes. They are replaced with an error 400 being returned to the malfiormed timeline request.

So with 0.9.16.6 the server will still crash if the client app is not changing how the timeline is constructed

saivajr, 2016-06-08:
So using Plex media Server 0.9.17 fixes the issue or not?
I don't have a Plex Pass so I have to wait for 0.9.17 to go public.

sa2000a, 2016-06-08:
The insult is that we have what PMS says is a malformed request from PHT. The handling of this changes in 0.9.17.0 in that an error response is sent back to PHT (Rasplex) instead of crashing - the impact will be that no timeline will be processed for watched state / progress

Fixing the request syntax at the PHT (Rasplex) would resolve the issue completely

NedtheNerd, 2016-06-17:
Can you check in newly released 1.6.1 and advise.

Helza, 2016-06-17:
Awesome, issue seems to be fixed

closed 2016-06-17 by Helza