RasPlex/RasPlex #493: Black screen when attempting playback

GiovanniPorta, 2016-05-15:
Raspberry Pi 2 Model B
RasPlex 1.0.3
32GB SD card

Every time I try to play a movie from my plex server, it just goes to a black screen indefinitely. Playing from Plex Web works perfectly fine. In addition, it doesn't even show the playback being started on the serverside log. I've tried reinstalling several times but the issue still persists.

g1t-dlanor, 2016-05-15:
Since movie playback works perfectly for a lot of people using the same RasPlex version on the same RPi2 model, something must be wrong at your end, though we still need to identify what that is.
As we still now very little about your setup, there are several questions we need answers to:

1: What is the type and class of your SD card (if in doubt, try showing us a picture of it) ?
2: What is the type and current capacity of your RPi2 power supply ?
3: How is your RPi2 connected to its LAN, and is that the same LAN as for the PMS server ?
4: Is your PMS signed in to Plex Inc, and shown as remotely connectable in Plex/Web settings ?
5: Is your RasPlex on RPi2 signed in to Plex  Inc, or just using local server discovery ?

Trying a different SD card and/or a stronger PSU (same voltage, but more current capacity) are two of the first things to try when having RPi problems in general.

Btw: A 32 GB SD card is unnecessarily large for RasPlex, which will work fine even on a 4 GB card.
Just make sure that the card is Class 10 and/or UHS-1, as lower class cards often give problems.

GiovanniPorta, 2016-05-19:
SD card ended up being the issue, moved to a new one and everything works fine. Weird because it _was_ a class 10, but it may have had other issues.

Thanks!

closed 2016-05-19 by GiovanniPorta