RasPlex/RasPlex #500: Crash when change Screen fps

Massaguana, 2016-06-07:
Here are an issue that occures with Rasplex 1.0.3 and also with new OpenPHT 1.6.0 for an RaspberryPi 3 with RemotePi Board.

When i Change the screen fps from 60 to 50 the system Crash´s...

Crash Report: 0607-1413
[image]

i think the [u] and [/u] are also wrong...

Massaguana, 2016-06-08:
some times RasPlex son´t save the fps Setting 50Hz. It Start Up with 60 HZ. Change is only to 59,94Hz possible, after an Restart is 50Hz back...

NedtheNerd, 2016-06-08:
I have tried to replicate your crash without success, changing refresh rate sticks on restart, closing issue.

closed 2016-06-08 by NedtheNerd

Massaguana, 2016-06-08:
Sorry Ned, but why there are an CrashReport? There are no Informations in it?

Here no problem to replictae this issue. not at all time but 25% Crash