RasPlex/RasPlex #501: force transcoding of HEVC video if platform Raspberry Pi

OttoKerner, 2016-06-07:
OpenPHT 1.6.0
Raspberry Pi 2

When playing 1080p HEVC encoded video, the playback is jerky. Which is no wonder since the RasPi cannot decode HEVC video in hardware and therefore tries to do it with its cpu.

OpenPHT should be aware of the limitations of the hardware it runs on, so it can tell PMS to transcode HEVC to AVC.

Kwiboo, 2016-06-07:
It should only direct play up to 720p HEVC encoded videos with default settings.

You can add a `advancedsettings.xml`-file with `<advancedsettings><codecs><video>-hevc</video></codecs></advancedsetting>` to fully disable hevc and tell PMS to transcode the video.

OttoKerner, 2016-06-07:
I have this 1080p video in my test library, and it Direct Played it.
```
<Media videoResolution="1080" id="682123" duration="2515440" bitrate="2638" width="1920" height="1080" aspectRatio="1.78" audioChannels="6" audioCodec="ac3" videoCodec="hevc" container="mkv" videoFrameRate="PAL" videoProfile="main">
<Part accessible="1" exists="1" id="683430" key="/library/parts/683430/1416060280/file.mkv" duration="2515440" file="G:\Test\The.Walking.Dead.Uncut.Season5Episode1.mkv" size="829328751" container="mkv" deepAnalysisVersion="1" indexes="sd" requiredBandwidths="3094,2622,2622,2622,2622,2622,2622,2622" videoProfile="main">
<Stream id="852713" streamType="1" default="1" codec="hevc" index="0" bitrate="2009" language="English" languageCode="eng" bitDepth="8" chromaSubsampling="4:2:0" codecID="V_MPEGH/ISO/HEVC" colorRange="tv" duration="2515440" frameRate="25.000" frameRateMode="cfr" height="1080" pixelFormat="yuv420p" profile="main" requiredBandwidths="2518,2048,2048,2048,2048,2048,2048,2048" width="1920"/>
<Stream id="852714" streamType="2" default="1" codec="ac3" index="1" channels="2" bitrate="192" language="en" languageCode="en" audioChannelLayout="stereo" bitDepth="16" bitrateMode="cbr" codecID="A_AC3" dialogNorm="-31" duration="2515440" requiredBandwidths="192,192,192,192,192,192,192,192" samplingRate="48000"/>
<Stream id="852715" streamType="2" selected="1" codec="ac3" index="2" channels="6" bitrate="384" language="en" languageCode="en" audioChannelLayout="5.1(side)" bitDepth="16" bitrateMode="cbr" codecID="A_AC3" dialogNorm="-31" duration="2515440" requiredBandwidths="384,384,384,384,384,384,384,384" samplingRate="48000"/>
<Stream id="1197553" key="/library/streams/1197553" streamType="3" selected="1" codec="idx" subIndex="2" language="Español" languageCode="spa" format="vobsub"/>
<Stream id="1286195" key="/library/streams/1286195" streamType="3" codec="idx" subIndex="1" language="Français" languageCode="fre" format="vobsub"/>
<Stream id="1286196" key="/library/streams/1286196" streamType="3" codec="idx" subIndex="0" language="English" languageCode="eng" format="vobsub"/>
<Stream id="1286197" key="/library/streams/1286197" streamType="3" codec="idx" subIndex="3" language="Nederlands" languageCode="dut" format="vobsub"/>
<Stream id="1286198" key="/library/streams/1286198" streamType="3" codec="idx" subIndex="5" language="Norsk" languageCode="nor" format="vobsub"/>
<Stream id="1286199" key="/library/streams/1286199" streamType="3" codec="idx" subIndex="4" language="Svenska" languageCode="swe" format="vobsub"/>
<Stream id="1286200" key="/library/streams/1286200" streamType="3" codec="idx" subIndex="6" language="Dansk" languageCode="dan" format="vobsub"/>
<Stream id="1286201" key="/library/streams/1286201" streamType="3" codec="idx" subIndex="7" language="Suomi" languageCode="fin" format="vobsub"/>
</Part>
</Media>
```

Kwiboo, 2016-06-07:
I see now that there is a small typo in the code at https://github.com/RasPlex/OpenPHT/blob/openpht-1.6/plex/Client/PlexTranscoderClientRPi.cpp#L153 preventing the 720p check from working, will be fixed in next point release, thanks!

NedtheNerd, 2016-06-16:
Fixed in 1.6.1, closing issue.

closed 2016-06-16 by NedtheNerd