RasPlex/RasPlex #503: Screensaver not displaying photos

Dan1001, 2016-06-09:
Hi,

I have the latest RasPlex 1.6.0. I'm having difficulty getting the screensaver to display my photos. The photos are in the Rasplex "Pictures" folder, but whilst the screensaver works fine showing fanart, if I select "photos" I just get a blank screen. Is this feature bugged, or am I doing something wrong?

many thanks,

Dan

g1t-dlanor, 2016-06-09:
I believe you may have misunderstood how this is supposed to work. The 'Photos' referred to are not just a collection of picture files in a local folder, but the "Photos" section as defined for the Plex server. Thus every instance of "OpenPHT" or "RasPlex" (which are now the same program on different platforms) will use the same set of photos, as distributed to them by the PMS server. There is no way (and no need) to define a separate set of screensaver pictures for each Plex client.

The "Pictures" folder as found in the RasPlex directory structure is a remnant from XBMC/Kodi storage structure. A lot of that structure is kept intact though currently unused by RasPlex.

Dan1001, 2016-06-10:
Thanks for looking at this. That's how I thought it would work initially - I have photos on the Plex server, and I can view them in Rasplex, but they don't show up in the screensaver. I assumed that was because I instead needed to put them in the Pictures folder, but it sounds like that was a wrong assumption.

So in fact I was mis-describing the bug: it's that photos on the Plex server don't work in the screensaver.

Dan

g1t-dlanor, 2016-06-10:
That's very strange, because I have no problem making the screensaver use my 'Photos' section instead of general 'Fanart'. The only issue I've had with it is that when I used the 'Preview' command in the screensaver settings menu, the very first picture fetch attempt has often failed. But after a short delay the screensaver would proceed to try another picture which always succeeded. And I only had that minor issue in older versions. With RasPlex 1.6.0 it works fine here even for the very first picture.

NedtheNerd, 2016-06-14:
A number of users are reporting this issue, including myself, it is being investigated, possible problem with PMS, not just the latest release but this has been happening for some time.

Dan1001, 2016-06-14:
cool - thank you. Yes, I've had this in all previous releases (and my bad for not mentioning it sooner)

Baschtlwaschtl, 2016-06-19:
Hi, i've updated to Raplex 1.6.1 and this problem still exists.. Maybe PMS problem...

danikuci1, 2017-12-01:
did this ever get resolved? i'm experiencing the same issue.

running rasplex version 1.8.0