RasPlex/RasPlex #504: SD card corruption on Raspberry pi 3

antonin42, 2016-06-09:
Using a rbpi 3, installation goes well, but at first reboot, sd card is always corrupted "beyond automatic repair". Tried 3 sd cards (2Go, transcend), same result on each.

NedtheNerd, 2016-06-10:
Use another SD card, check your computer, not a Rasplex issue, closing.

closed 2016-06-10 by NedtheNerd

g1t-dlanor, 2016-06-10:
@antonin42:
I agree with @NedtheNerd that this is not an issue with RasPlex. If it were then there would be a lot more users with the same problem. But I think you deserve a little more guidance to make it work.

'Traditionally' there are two main causes for SD card corruption reported by RasPlex users:

First and foremost is the use of SD cards which are lower class than 10 (even if marked as Class10).
This is easily tested simply by flashing the SD card on a fast computer with "Win32 Disk Imager" or with the RasPlex installer. Speeds below 10 MB/s indicate lower class cards. Unfortunately there are a LOT of cards on the market which are falsely marked as Class10. (I recently bought a few such cards myself, which in fact turned out to be Class4, though clearly marked as Class10 and even UHS-1 with the proper logos for each of these standards.)

Second is the use of a power supply that suffers from any of 3 distinct problems:
2a: Too low current capacity to deliver a steady 5 volts
2b: Bad design causing unfiltered voltage spikes on the 5 volt output
2c: Inappropriate turnoff without proper RasPlex shutdown (can break power during log writes)

Both cases 2a and 2c are quite common among people who power the RPi from the USB ports of a TV set, and without proper CEC command support to tell  RasPlex about an impending turn-off.