RasPlex/RasPlex #505: No longer possible to pause during playback with P

RH0M8U5, 2016-06-10:
Hi, 
on last day I updated my rasplex to 1.6.0 from 1.0.3.
On 1.0.3 all works fine. I could start with "P" the playback directly from overview, and during the playback it was possible to pause.
With 1.6.0 its no longer possible, only to start the media from overview.
I looked at the changes and fixes, but there I found no change for it.
So, I think its a bug or something like that.

I am using a Flirc Dongle + XboxOne Remote

Kind Regards

NedtheNerd, 2016-06-10:
Can you try using CEC (TV remote), disconnect the FLIRC dongle first.

Could be that the XBoxOne Remote mappings are now only mapping Play and not PlayPause.

RH0M8U5, 2016-06-10:
Yes, I have and I tried now.
I have a LG TV and there on the reYes, I have and I tried now.
I have a LG TV and there on the remote is play and pause for each a own button
and not on the same button like on the Xbox remote.
But there works pause and play, but when a push play during playback nothing happened. 

I have used a second micro sd-card  and a flashed 1.0.3 on it, and there "P" normal works with Flirc and Xbox remotemote is play and pause each a own button
and not on the same button like on the Xbox remote.
But there works pause and play, but when a push play during playback nothing happened. 

I have used a second micro sd-card  and a flashed 1.0.3 on it, and there "P" normal works with Flirc and Xbox remote

NedtheNerd, 2016-06-10:
That's because on CEC play is Play and pause is Pause, neither is PlayPause.

As I said it could be that the mappings have changed, you can create a keymap.xml to remap the play button back to playpause.

RH0M8U5, 2016-06-10:
First I wanted to ask for help, because that keymaps.xml thing looks weird to me.
But than I found an alternative way to my problem. It looks like, that "SPACE" makes now the same, like "P" before.
Now I only need to change my Flirc config

closed 2016-06-10 by RH0M8U5

NedtheNerd, 2016-06-10:
Glad you have a solution.