RasPlex/RasPlex #507: GUI Freeze While OMX.broadcom.video_decode - OMX_ErrorInsufficientResources, insufficient resources

benjaminwolf, 2016-06-16:
Had a GUI freeze up while attempting to play a file. Was able to SSH in and snatch the logs and reboot from the software, so only the GUI was unresponsive. Appears to have entered a loop condition when the Video Decoder ran out of resources. See attached log excerpt.

I've included the media information below. The item played fine immediately after a reboot. Rasplex/OpenPHT had been working fine for days without reboot before this loop condition.

HW: rPi3 w/ 2.5a 5v Powersupply
SW: Rasplex/OpenPHT 1.6.0
PMS: 0.9.17.2.2159-2bd156c :: Windows 2012 R2
ENV: Local PMS hardlined to rPi3 hardline connection.

[plexhometheater.zip](https://github.com/RasPlex/RasPlex/files/317674/plexhometheater.zip)
[pht_excerpt.txt](https://github.com/RasPlex/RasPlex/files/317675/pht_excerpt.txt)

[PMS Media Info]

> <MediaContainer size="1" allowSync="1" identifier="com.plexapp.plugins.library" librarySectionID="2" librarySectionTitle="TV Shows" librarySectionUUID="2860eedf-2eb3-47c7-b9c8-2124a07b9b01" mediaTagPrefix="/system/bundle/media/flags/" mediaTagVersion="1465523274"><Video ratingKey="2527" key="/library/metadata/2527" parentRatingKey="2526" grandparentRatingKey="1327" guid="com.plexapp.agents.thetvdb://121361/1/1?lang=en" librarySectionID="2" type="episode" title="Winter Is Coming" grandparentKey="/library/metadata/1327" parentKey="/library/metadata/2526" grandparentTitle="File Name" contentRating="TV-MA" summary="blah blah blah blah" index="1" parentIndex="1" rating="7.9" viewCount="16" lastViewedAt="1465987390" year="2011" thumb="/library/metadata/2527/thumb/1465784430" art="/library/metadata/1327/art/1465784437" parentThumb="/library/metadata/2526/thumb/1465784436" grandparentThumb="/library/metadata/1327/thumb/1465784437" grandparentArt="/library/metadata/1327/art/1465784437" grandparentTheme="/library/metadata/1327/theme/1465784437" duration="3697155" originallyAvailableAt="2011-04-17" addedAt="1347790758" updatedAt="1465784430" chapterSource=""><Media videoResolution="1080" id="128651" duration="3697155" bitrate="10160" width="1920" height="1080" aspectRatio="1.78" audioChannels="6" audioCodec="dca" videoCodec="h264" container="mkv" videoFrameRate="24p" audioProfile="dts" videoProfile="high"><Part accessible="1" exists="1" id="177653" key="/library/parts/177653/1427046300/file.mkv" duration="3697155" file="D:\0\medianame.mkv" size="4695214941" audioProfile="dts" container="mkv" indexes="sd" videoProfile="high"><Stream id="1254048" streamType="1" default="1" codec="h264" index="0" bitrate="8648" language="English" languageCode="eng" bitDepth="8" cabac="1" chromaSubsampling="4:2:0" codecID="V_MPEG4/ISO/AVC" duration="3697155" frameRate="23.976" frameRateMode="cfr" hasScalingMatrix="0" headerStripping="1" height="1080" level="40" pixelFormat="yuv420p" profile="high" refFrames="4" scanType="progressive" width="1920"/><Stream id="1254049" streamType="2" selected="1" default="1" codec="dca" index="1" channels="6" bitrate="1509" language="English" languageCode="eng" audioChannelLayout="5.1(side)" bitDepth="24" bitrateMode="cbr" codecID="A_DTS" duration="3697152" profile="dts" samplingRate="48000"/><Stream id="1254050" streamType="3" codec="srt" index="2" language="English" languageCode="eng" codecID="S_TEXT/UTF8" format="srt"/><Stream id="1445573" key="/library/streams/1445573" streamType="3" codec="srt" language="English" languageCode="eng" format=""/><Stream id="1439400" key="/library/streams/1439400" streamType="3" codec="srt" language="English" languageCode="eng" format=""/><Stream id="1439401" key="/library/streams/1439401" streamType="3" codec="srt" language="English" languageCode="eng" format=""/></Part></Media><Writer id="0000" tag="Nobody"/><Writer id="00000" tag="Nobody"/><Director id="00000" tag="Nobody"/><Extras size="0">
> </Extras></Video></MediaContainer>

[MediaArea.net/MediaInfo]

> General
> Unique ID                                : 
> Complete name                            : D:\0\medianame.mkv
> Format                                   : Matroska
> Format version                           : Version 4 / Version 2
> File size                                : 3.71 GiB
> Duration                                 : 49mn 58s
> Overall bit rate                         : 10.6 Mbps
> Encoded date                             : UTC 2016-04-25 04:20:12
> Writing application                      : mkvmerge v7.0.0 ('Where We Going') 64bit built on Jun 22 2014 08:19:39
> Writing library                          : libebml v1.3.0 + libmatroska v1.4.1
> 
> Video
> ID                                       : 1
> Format                                   : AVC
> Format/Info                              : Advanced Video Codec
> Format profile                           : High@L4.1
> Format settings, CABAC                   : Yes
> Format settings, ReFrames                : 4 frames
> Codec ID                                 : V_MPEG4/ISO/AVC
> Duration                                 : 49mn 58s
> Bit rate                                 : 10.0 Mbps
> Width                                    : 1 920 pixels
> Height                                   : 1 078 pixels
> Display aspect ratio                     : 16:9
> Frame rate mode                          : Constant
> Frame rate                               : 23.976 (24000/1001) fps
> Color space                              : YUV
> Chroma subsampling                       : 4:2:0
> Bit depth                                : 8 bits
> Scan type                                : Progressive
> Bits/(Pixel*Frame)                       : 0.202
> Stream size                              : 3.50 GiB (94%)
> Writing library                          : x264 core 148 r2665+53 6629127 t_mod_Custom [8-bit@all X86_64]
> Encoding settings                        : cabac=1 / ref=4 / deblock=1:-3:-3 / analyse=0x3:0x113 / me=umh / subme=11 / psy=1 / fade_compensate=0.00 / psy_rd=1.00:0.00 / mixed_ref=1 / me_range=32 / chroma_me=1 / trellis=2 / 8x8dct=1 / cqm=0 / deadzone=21,11 / fast_pskip=0 / chroma_qp_offset=-2 / threads=24 / lookahead_threads=4 / sliced_threads=0 / nr=0 / decimate=0 / interlaced=0 / bluray_compat=0 / constrained_intra=0 / fgo=0 / bframes=8 / b_pyramid=2 / b_adapt=2 / b_bias=0 / direct=3 / weightb=1 / open_gop=0 / weightp=2 / keyint=250 / keyint_min=23 / scenecut=40 / intra_refresh=0 / rc_lookahead=60 / rc=crf / mbtree=0 / crf=17.0000 / qcomp=0.60 / qpmin=0:0:0 / qpmax=69:69:69 / qpstep=4 / vbv_maxrate=62500 / vbv_bufsize=78125 / crf_max=0.0 / nal_hrd=none / filler=0 / ip_ratio=1.40 / pb_ratio=1.30 / aq=3:0.80 / aq-sensitivity=10.00 / aq-factor=1.00:1.00:1.00 / aq2=0 / aq3=0
> Default                                  : Yes
> Forced                                   : No
> Color range                              : Limited
> Matrix coefficients                      : BT.709
> 
> Audio
> ID                                       : 2
> Format                                   : AC-3
> Format/Info                              : Audio Coding 3
> Mode extension                           : CM (complete main)
> Format settings, Endianness              : Big
> Codec ID                                 : A_AC3
> Duration                                 : 49mn 58s
> Bit rate mode                            : Constant
> Bit rate                                 : 384 Kbps
> Channel(s)                               : 6 channels
> Channel positions                        : Front: L C R, Side: L R, LFE
> Sampling rate                            : 48.0 KHz
> Frame rate                               : 31.250 fps (1536 spf)
> Compression mode                         : Lossy
> Stream size                              : 137 MiB (4%)
> Default                                  : Yes
> Forced                                   : No

NedtheNerd, 2016-06-17:
Can you check in newly released 1.6.1 and advise.

benjaminwolf, 2016-06-17:
Don't know if I can force the condition to happen. I ran the same file again, but without getting back into the condition. That's to be expected though, as after rebooting the PI the first time, it did not fall into the same condition. Not sure this is something that can be tested for easily.

NedtheNerd, 2016-06-17:
OK, thanks, I'll close this for now, if it happens in the future please feel free to reopen with further comment.

closed 2016-06-17 by NedtheNerd