RasPlex/RasPlex #508: OPHT 1.6.0 - Skip ahead results in frozen video

trumpy81, 2016-06-18:
**RPi model**: RPi2 Model B
**RasPlex**: OpenPHT 1.6.0
**PMS**: 0.9.17.2 on Synology DS1813+
**Network**: WiFi

When reaching the end of a TV episode, I often use skip ahead which normally, will jump ahead by 30secs. However, if I use skip ahead and there is less than 30secs remaining, the video will simply stop and nothing else happens, the video does not end until I press the stop button or the exit button.

Any ideas why this would happen, it did not happen in RasPlex 1.0.3 or earlier.

NedtheNerd, 2016-06-18:
Adaptive seeking is now used which changes the behaviour, if you wish to go back to the old way of 'jumping' then just go to Preferences-Change Preferences for Playback-Advanced video and set the 'Skip delay' to 0.

Closing issue

trumpy81, 2016-06-19:
Thanks Ned, but your suggestion does NOT fix the issue at all. 

Regardless of the skip delay setting (the default is 750ms but is now set to none), when there is less than 30 secs of video remaining and you use the skip ahead button, instead of the video ending, it simply freezes and will not do anything as though it is waiting for more video or as if there is no end of file trigger.

Pressing stop or exit is the only way to get the file to stop playing and to return to the pre-play screen.

If the end of the file is reached it should stop playback and return to the pre-play screen as it does with RP 1.0.3.

Just to be more clear, by skip ahead I mean using the left/right direction keys (either side of the OK button) on the remote, not the FF or Rewind buttons.

Keep in mind that I am using the RemotePi and a Logitech Harmony Ultimate One remote with the default Windows Media PC device configuration.

NedtheNerd, 2016-06-19:
Debug log file please.

trumpy81, 2016-06-19:
I played an episode of Castle and at the end, skipped ahead twice and the video locked up as usual. I let it sit for about a minute then captured the logs.

Time was around 2:40am.

[PHTLogs.zip](https://github.com/RasPlex/RasPlex/files/322400/PHTLogs.zip)

trumpy81, 2016-06-28:
After installing 1.6.2 I am still experiencing this issue. I have attached the logs taken after the latest incident of this issue.

I let PHT sit in the locked condition for a minute or so while I grabbed the logs.

[OPHTLogs.zip](https://github.com/RasPlex/RasPlex/files/338233/OPHTLogs.zip)