RasPlex/RasPlex #509: Clock and Movie/TV title cut off on screensaver overlay

jamesmacwhite, 2016-06-18:
After calibrating my LG TV (37LS5600) via the video calibration tool, I've noticed that the screensaver clock and movie/tv title overlay are cut off to the left and right side depending on where it appears in the corners. This is using the default skin.

**RPi model**: 2
**RasPlex**: 1.6.0.113-46fadd5e
**PMS**: 0.9.16.6 (Stable) CentOS 7
**Network**: Ethernet

It appears to be isolated to this specific TV as I have another RasPlex device using the same resolution and refresh rate settings that doesn't have this problem, but it is a completely different brand and model of TV.

Any ideas on what I can do to fix it?

Kwiboo, 2016-06-18:
This is most likely due to overscan by your TV, there is usually a "pc mode", "dot by dot" or similar setting on your TV to show the complete picture, there should normally not be any need to use the video calibration if your TV is configured to show the complete picture.

NedtheNerd, 2016-06-18:
Find out what modes your TV's supports by running tvservice --modes=CEA, the results will return a (prefer) one that it will default to, if this is not appropriate you can force a different mode using the info here https://www.raspberrypi.org/documentation/configuration/config-txt.md

As this is not a Rasplex issue I am closing it.  It would have been better to raise this on the forum where you would get responses from the entire Rasplex community.

closed 2016-06-18 by NedtheNerd