RasPlex/RasPlex #511: [Feature Request] ability to use touchscreens on rasplex.

Bdavie82, 2016-06-21:
Model: I am using a Raspberry Pi2
Rasplex version: Any
PMS: 2.6.1 on Windows 10 Pro
Network: Wifi

I recently got my Raspberry Pi2 running Rasplex 1.6.1. Things seem to work fine in general. I would like to be able to use a little 5' touchscreen that I got along with the Pi in order to make a little media hub. however I am finding that there is no way to install the needed drivers for the screen to even be used as a display let alone the touch portion to be used for navigation. I would like to see a feature added that allows for these two drivers to be added to RasPlex so that the touchscreen can be used as the input device. I am happy to provide any additional information I can for this. Thank you.

iancrosser, 2016-08-15:
I too would find the ability to use the touch screen on my 7" Pi Foundation screen very useful.  The screen works, however having to rely on a plugged in USB device or a WiFi router to connect my phone as a remote is not the most ideal for the tablet like experience I'm going for.

shuguet, 2016-09-10:
Another idea to go along would be to allow RasPlex to manage an USB sound card to output music to a sound system independently of the TV. That way RasPlex could be used as a Media Center to play videos and music on the TV via HDMI. Or used as a sound system only (connected to a sound bar for exemple) using the touchscreen as control.

kirklandia, 2017-07-04:
I am making car clients for road trips in lieu of phone or tablet devices.  I would like touch screen support also as the whole point (for me) is to not have phone and tablet devices.
If there's a way to hack this together unofficially, I'd be interested in seeing it.

silverbullettruck2001, 2017-08-26:
Has there been any progress in making the touchscreen work with RasPlex yet?

silverbullettruck2001, 2017-09-04:
@kirklandia did you ever figure out a way to get the touchscreen to work on RasPlex?