RasPlex/RasPlex #512: OPHT 1.6.1 - Frozen at startup

trumpy81, 2016-06-22:
**RPi model**: RPi2
**RasPlex**: 1.6.1.120
**PMS**: 1.0.0 Beta
**Network**: WiFi

Started the RPi2 this evening and after displaying the OpenPHT logo the RPi2 froze and did not continue booting. It seemed responsive and I was able to SSH into it, but the on screen image did not advance past the OpenPHT logo.

Looking at the logs it would appear to have been an issue with CEC. Logs Attached.

[OPHTLogs.zip](https://github.com/RasPlex/RasPlex/files/327805/OPHTLogs.zip)

NedtheNerd, 2016-06-27:
Can you load latest release 1.6.2, there are a number of fixes for issues caused by PMS 1.0.0.

trumpy81, 2016-06-28:
I have hammered 1.6.2 for a day or two now and so far I have not seen a repeat of this issue. I'm still having the other issue though.

closed 2016-06-29 by NedtheNerd