RasPlex/RasPlex #513: Rasplex 1.6.1 - channels not working when using IOS remote

tropas2001, 2016-06-25:
[Tell us about your issue, steps to reproduce, expected and actual behavior]
[Always attempt to reproduce your issue in the default skin]
[Enable debug logging in preferences and insert a link to your log-file]

**RPi model**: [raspberry pi model]
**RasPlex**: [version and skin]
**PMS**: [version and operating system]
**Network**: [ethernet/wifi]

None of the BBC iPlayer channels and ITV player channels will work when being controlled by the remote, but will work when using a keyboard. An error comes up on the remote "couldn't retrieve the play queue for this item. Please try again". Also, I've tried to use the API over the JSON interface and Rasplex fails to respond properly (i.e. return data)
RPI3
plexmediaserver Version 0.9.16.6 on debian
Ethernet
Plex IOS app - 4.1.2

ToLoKn, 2017-01-06:
Hi there, I'm having a very similar problem with the new Rasplex release.
When trying to play audio or video stream from the "ARD Mediathek 2016" channel playback on Rasplex starts. After a frew seconds the iOS app reports "Die Warteschlange für diesen Titel konnte nicht aufgerufen werden. Probieren Sie es noch einmal" (which is the locaclized error message "Couldn't retrieve the play queue for this item. Please try again") and playback stops.
Start playing the channels from the web interface of PlexMediaServer as well as using keyboard works fine.
But once the Rasplex player is chosen in the IOS app, playback stops with the same error message.

Setup is:
plexmediaserver Version 1.3.3.3148 on ubuntu
RPI3 with RasPlex (official) Version: 1.7.1.137-b604995c
Ethernet and/or WLAN
Plex IOS App 4.6.1

tropas2001, 2017-02-05:
Hi - is there an update for this issue? is this an IOS app issue or rasplex issue?