RasPlex/RasPlex #514: PLaylist not showing up after server update

rubix45, 2016-06-25:
I just updated my plex server to Version 1.0.0.2261 and when i went to my rasplex devises, either running Version 1.6.1 or Version 1.0.3, none of my playlists would show up.  

I tried looking for settings on the server and on the Rasplex device.  i have also tried logging out of the device and logging back in.    I also tried creating a new playlist but that one did not show up either.

Anyone got any ideas?

Kwiboo, 2016-06-25:
This is a known issue and are being worked on, see forums.plex.tv/discussion/222663/important-information-playlists-broken-with-pms-1-0-0/p1

rubix45, 2016-06-25:
Thank you Kwidoo.  I will look for the update when it comes.  you are AWESOME!!

NedtheNerd, 2016-06-26:
Fixed in next release - 1.6.2, closing issue.

closed 2016-06-26 by NedtheNerd