RasPlex/RasPlex #516: Rasplex reboots during setup when CEC device connects

CookieMonger, 2016-07-03:
Upon initial setup of RasPlex, as soon as CEC is detected and connected the system becomes unresponsive and reboots.  I am not able to control anything at all before this occurs.  The TV connected is a Vizio M55-C2.

**RPi model**: RPI3
**RasPlex**: 1.0.3 & 1.6.2

g1t-dlanor, 2016-07-03:
I suggest looking through all of the configuration menus of the TV set, to see if you can disable some CEC usage which is causing this problem. Different TV manufacturers have implemented the protocol with their own variations, and while RasPlex is designed to work with most of them, yours apparently has some variation unsuitable for RasPlex. Possibly it's just some single feature of the TVs CEC usage that needs to be disabled to make it work properly. But if you can disable all of it, that would be the quickest way to get RasPlex up and running. Then you can configure the CEC usage of RasPlex to not allow the TV set to shut down RasPlex in any circumstance, after which you can enable the CEC usage of the TV set again.

There are some possible complications to consider:
If you are powering the RPi3 from a USB port of the TV set, that in itself can cause serious problems, as TV USB ports are not designed to deliver sufficient current for an RPi3. And if the USB ports turn off with the TV set in standby mode, that may corrupt SD card contents of the RPi3.

Another complication is if you planned to use the TV remote to control RasPlex through CEC, which obviously won't work while CEC is disabled as I suggested doing above. So for such a case you may need to connect a USB keyboard instead, for initial configuration of RasPlex.

CookieMonger, 2016-07-04:
I was able to disable the reboot loop by blocking CEC shutdown as you suggested but now it's as if my TV is sending no commands at all.  The pi is completely unresponsive to my remote control commands despite recognizing that CEC is connected and enabled.  

I've tried the same TV and HDMI cable with Xbian and it works without any issues, though the plex client there runs terribly.  Any ideas would be greatly appreciated @g1t-dlanor.

CookieMonger, 2016-07-04:
It looks like Xbian is using libcec 3.1.0 and Rasplex is running 3.0.1

g1t-dlanor, 2016-07-04:
Unfortunately I'm not able to help you much further with this myself as my own TV sets don't use CEC and I'm not the one to make such coding changes (I'm mainly a beta-tester), but hopefully one of the the coding project members using CEC will attend to this issue soon.

I find it strange that disabling CEC shutdown should block all other commands though.
Please double check the RasPlex CEC adapter settings, to ensure that you haven't disabled the top entry in that list, which is intended to disable all CEC. For your problem it should be sufficient to disable just one or two entries further down in that list (or so I hope anyway...).

If that doesn't work (possibly what you already tried), then RasPlex will need coding changes to allow proper use of CEC with your TV set. So one of our programmers will have to attend the issue then.

slambert, 2016-07-12:
I'm seeing this same issue with a RPI B+ and a VIZIO M43-C1. Reflashed RasPlex to the SD card just to be sure something odd wasn't up and ended up here.

FWIW I saw a crash id of 0712-1816. By the way, the crash report suggests you go to tiny.cc/rasplex-bugs, but that URL doesn't connect to anything.

CookieMonger, 2016-07-13:
I was able to get the raspberry pi to respond to commands by doing the following:
turning my television input to the pi
turning off the tv
turning on the tv and spamming an input on my remote

It seems to recognize my remote until the pi reboots if I follow this process.

NedtheNerd, 2016-07-13:
Try changing the HDMI lead or HDMI input on your TV.

slambert, 2016-07-15:
Ok, @CookieMonger try this. I think it's working for me but I haven't tested it thoroughly... also assuming you have enough power going to the board.

1. Go to Preferences > System > Input Devices > Peripherals and hit OK
2. Select the CEC Adapetr for your TV - mine is `Product ID: 1001 Vendor: 2708 Class: cec Version: libCEC 3.0,1 - firmware v1` and hit OK
3. I kinda shut everything down from here. "When the TV is switched to off" I set to ignore. There's a few other settings there that control shutdown on the Pi and the TV. 

That basically worked for me **(though the remote still has moments of being unresponsive)**. If you need more help/specifics, let me know.

By the way, I had this working before, so maybe some of the settings were changed upon update?

slambert, 2016-07-15:
@CookieMonger What do you mean when you say "spamming an input on my remote"?

christopherdaly-wf, 2016-07-15:
I repeatedly press the up directional button on my remote while the tv is turning on.

The pi has a brand new 2.5A power supply and I've tried 2 hdmi cables that function perfectly with the same pi on another tv.

@slambert I'll forward your suggestions on to my parents and update when I hear back as I no longer have direct access to the device.

circulon, 2016-07-23:
I am getting the same thing ie no CEC throughput at all as far as i can tell ;(
I have turned on Debug logging as per @nedthenerd other post for tweaking cec commands
made sure cec is turned on in the tv
Tried different ports & cables

The TV is found and appears in the peripherals list.
Strangely sometimes I get an option to select the TV hdmi port and sometimes not
This appears directly above the HDMI port override.

Is there any other way of testing if the cec commands are making it through to the pi?
Running 
Rasplex 1.6.2 on Pi B+

fleehay15, 2016-12-13:
Same happened to me running rasplex 1.7.1 RPi2 on my Vizio 55" lcd. 
My fix, on the vizio tv:
menu > system > cec > disable ! (i dont use a remote, i cast using my plex app via the iphone)

http://www.howtogeek.com/207186/how-to-enable-hdmi-cec-on-your-tv-and-why-you-should/

entmike, 2017-04-14:
Same issue, 4k Vizio (don't have exact model) - Fresh SD with Rasplex, it crashes after detecting CEC.  If I disable it on the TV, no crashing.  Will see if I can disable it in Rasplex later, as I don't want to disable CEC on TV entirely.

slambert, 2017-04-14:
My current workaround is:

1. Turn on TV, remote will go unresponsive.
2. Change Input from Rasplex to something else (HDMI 3, or whatever)
3. Change input back to Rasplex

Then the remote works. It's annoying, but I still have one remote and it works. No idea how to fix it... but maybe someday a hero will arrive to this thread...

jimmyeao, 2017-05-23:
Having this same issue, unable to find a workaround. Is it possible to SSH into the PI and disable the function? It doesn't seem to matter if I disable or enable CEC on my TV, it is always detected and the PI always reboots...