RasPlex/RasPlex #518: IR Remote Control stops working after upgrade to version 1.6.2

chuckofthenorth, 2016-07-11:
I've been using an Xbox IR remote control on my RasPlex since I built it in March (using 1.0.3).  Worked perfectly.  After the software upgrade last week to 1.6.2, the IR remote stopped working.

I did some testing today and have confirmed that the remote works fine with a clean flash back to RasPlex 1.0.3 and does not work with a clean flash to RasPlex 1.6.2.   Everything else seems OK (I can control the box with a keyboard or IOS app on the network), but not via the IR remote since the RasPlex upgrade.

Hardware Details: 
 - Raspberry Pi 3
 - Flash Card: 16GB Adata microSDHC (Class 10) for both configurations.
 - Remote: USB attached XBOX IR Receiver with remote XBOX IR control

RasPlex Version that works OK:  RasPlex-RPi2.RP-1.0.3.img.gz
RasPlex Version that IR fails in: RasPlex-1.6.2.123-e23a7eef-RPi2.arm.img.gz

I've ssh'ed to the device and looked in /var/logs, but don't see anything interesting.  Running ps | grep lirc reveals the following processes are running:

ps | grep lirc
  384 root       0:00 /usr/sbin/eventlircd -f --evmap=/etc/eventlircd.d --socket=/run/lirc/lircd --release=_UP
  491 root       0:00 /usr/sbin/lircd --uinput --driver=default --device=/dev/lirc0 --output=/run/lirc/lircd-lirc0 --pidfile=/run/lirc/lircd-lirc0.pid /etc/lirc/lircd.conf.rpi
  529 root      15:26 /usr/lib/plexht/plexhometheater --standalone -fs --lircdev /run/lirc/lircd
  678 root       0:00 grep lirc

I'm not (yet) sure how to troubleshoot lirc in Linux, but if someone gives me some pointers, I can ssh to the device and dig out more info.

**RPi model**: [Pi3]
**RasPlex**: [1.6.2 Default skin]
**PMS**: [Version 1.0.0.2261 on Arch Linux 64]
**Network**: [ethernet]

NedtheNerd, 2016-07-13:
/var/logs is the incorrect place to look, please look in /.plexht/temp, you will find plexhometheater.log.

You will need to turn on debug in Preferences then restart the RPi, any button presses or errors should then be found in the log file.

You can also run:

     tail -F /storage/.plexht/temp/plexhometheater.log

This will give you a real time view of the log in an SSH session, with this running try pressing buttons and entries should appear.

chuckofthenorth, 2016-07-14:
Thanks for the guidance.  I have uploaded my plexhometheater.log log file at [http://pastebin.com/0Y36FWSV](http://pastebin.com/0Y36FWSV). 

- No key-presses from the remote were detected, despite many different keys and numerous attempts.  
- I unplugged and re-inserted the USB receiver for the IR (~ 28 minute mark) to see if it would discover the IR remote and start working.  No change.
- At 37:22, I plugged in a physical USB keyboard and those key-presses are in the log toward the end.  But still nothing from the IR.   

Of note:  After one reboot cycle (of dozens), the IR worked - until the next reset.  Unfortunately, that was before I turned on the debug logs.  If I can get a capture with it working, I'll upload so we can compare what is different.

chuckofthenorth, 2016-07-14:
After numerous reboots, I good a good one.  Here is the [log](http://pastebin.com/0PJcGcEB) file.

In this case, the only thing different was a power off/on reset - The hardware was identical, and the configuration was unchanged (same USB ports used etc), but this time the IR remote worked perfectly.  Hopefully the delta will reveal something.

xmeunier, 2016-07-23:
I have the same issue with 1.6.2, since this morning my IR (USB) keyboard (iClever) is not seen anymore.

I tried several reboots and power on/off, all of them fails

I will put debugs logs soon. a lsusb with ou without the usb receiver plugged give the same output:

RasPlex:~ # lsusb
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

xmeunier, 2016-07-23:
Hum... i tried with a new SD card and a 1.6.1 and i still have the issue. My IR keyboard is working fine on MacOS so the issue is not here... Next try with raspbian

chuckofthenorth, 2016-07-25:
No such "over-current" messages in my log and my problem still exists.  RasPlex 1.6.2 remains unusable for me, so for now, I've had to abandon the RasPlex on my Pi and switch my wife back to the Android media box.  The earlier RasPlex version worked so well.  I hope someone figures out what the root cause and fix is is before the next release.

xmeunier, 2016-07-26:
Way you don't come back to the previous version 1.6.1?

chuckofthenorth, 2016-07-27:
OK.  I flashed back to 1.6.1.  Problem is still there, but maybe occurs less frequently.  IR Remote works 1 in 7 or 8 tries, rather that 1 in 12 to 15.  So the bug or race condition has been in there since then. 

I've looked at the [logfile](http://pastebin.com/0Y36FWSV) I posted 2 weeks ago, and still don't see anything wrong.  Perhaps expert eyes would spot something.

1lostsock, 2016-07-27:
Just upgraded to 1.6.2 a couple of days ago and have this same issue with the original XBox DVD remote.

MKParis, 2016-09-30:
I bought a usb dongle FLIRC. I set it on my mac and that worked very well in 1.6.1. 
Since 1.6.2 if I let FLIRC connected, then i the following behavior: 
it is as if someone press all the buttons simultaneously. 
I have to let FLIRC disconnected and remotely control Rasplex with my iPad.