RasPlex/RasPlex #521: Feature Request - Sync

clefus, 2016-07-19:
Rasplex should add sync to their list of features.  This would set it apart from many of the existing options out there....it could be highly effective and useful in a variety of situations.

Thanks for reading/considering.

[Tell us about your issue, steps to reproduce, expected and actual behavior]
[Always attempt to reproduce your issue in the default skin]
[Enable debug logging in preferences and insert a link to your log-file]

**RPi model**: [raspberry pi model]
**RasPlex**: [version and skin]
**PMS**: [version and operating system]
**Network**: [ethernet/wifi]

Ruvetuve, 2016-07-20:
If it can sync to an attached Hard Disk, this feature would be great for viewing Plex content on the road. Thinking about plugging it in on hotel TV's etc.

clefus, 2016-07-20:
I agree Ruvetuve.  While copying them to the SD card is possible....I think that sync should still be implemented.

It could also benefit those with network problems or slow wifi in their homes.  But, at the end of the day - the designers need to know that not all users of rasplex are savy enough to copy to SD etc.  (IE my wife lol).  I like the KISS principle.....keep it simple....

I truly believe this one feature would set RASPLEX out above some of the other players that are out there....and it would have many practical uses for people.  On the road for kids, hotels, portable kiosks, and so much more.

leprasmurf, 2016-08-02:
:+1: this feature would be immensely useful.  Especially for clients that rely on wifi.  The server can transcode for simplified playback before sync'ing, that removes the need for transcoding by the Pi.

ovizii, 2016-08-02:
I agree but unfortunately RasPlex is built upon Plex Home Theater (PHT) and PHT had no sync option so it looks like its not going to happen too soon.

vbtrek, 2018-04-08:
I'd like sync too, because I have an RPI stuck to the back of the TV with a USB HDD also stuck to the back of the TV which we use at home and whilst travelling in the caravan. Any other solutions available for RPI other than RasPlex?

WebSpider, 2018-04-08:
OpenPHT on raspbian with Plex Media Server on the same box should work

justinglow, 2018-09-27:
Would also love syncing locally for playing offline.