RasPlex/RasPlex #526: OpenVPN not connecting during boot using systemctl

tailorian, 2016-07-29:
I'm trying to automatically connect to my openvpn server during system start, right after the network service come online.

I'm using the method described in `/storage/.config/system.d/openvpn.service.sample`

But at system start the following messages are found in the log:
```
XXXX RasPlex systemd[1]: [[0;1;39m[/storage/.config/system.d/openvpn.service:X] Unknown lvalue 'Requires' in section 'Service'[[0m
XXXX RasPlex systemd[1]: [[0;1;39m[/storage/.config/system.d/openvpn.service:X] Unknown lvalue 'After' in section 'Service'[[0m
```
The value in Requires and After are:
```
Requires=network-online.service
After=network-online.service
```
But when I start the openvpn service manually it starts normally and works.
`systemctl start openvpn.service`

**RPi model**: Raspberry Pi 1B
**RasPlex**:  1.6.2.123-e23a7eef
**PMS**: 1.0.3.2461
**Network**: ethernet

tailorian, 2016-08-01:
A user in the [RasPlex forum](https://forums.plex.tv/categories/rasplex) found a solution

https://forums.plex.tv/discussion/comment/1186871/#Comment_1186871

The service config should look like this:
```
[Unit]
Description=openvpn
After=network-online.service
Requires=network-online.service
[Service]
Type=forking
ExecStart=/usr/sbin/openvpn --daemon --config /storage/.config/vpn-config/OVENVPNclientCONFIG.ovpn
TimeoutStopSec=1
Restart=always
RestartSec=2
StartLimitInterval=0
[Install]
WantedBy=multi-user.target
```

Maybe the example config should be updated!

g1t-dlanor, 2016-08-01:
If that is what works, then I agree that the script ought to be amended this way.

But this script is not really something the RasPlex devs have produced, but is merely inherited from the existing LibreELEC implementation that RasPlex uses. Thus it's really the LibreELEC coders that should change the script in their sources, so that it will be fixed for all applications based on LibreELEC, including both Kodi and RasPlex releases.

benjaminwolf, 2016-08-05:
This is my service and works perfectly. My full guide is located here: https://forums.plex.tv/discussion/comment/1186871/#Comment_1186871

I'll reach out to the LibreELEC team and see if they're interested in using it. I'd prefer to be credited for the contribution, other than 'Forum User' though.