RasPlex/RasPlex #527: GUI freeze

HappyCerberus, 2016-07-30:
RPi 2
Rasplex: 1.6.2.123-e23a7eef
PMS: Linux / long term issue
Network: wired

Recorded log:
http://pastebin.com/raw/bkfQtLL1

g1t-dlanor, 2016-08-01:
More details please...!
Reading just the logs won't tell us what you were trying to do with the GUI when it happened.
And that knowledge is crucial to any attempt to replicate the problem.

HappyCerberus, 2016-08-01:
It randomly freezes when browsing the videos, so it's kind of hard to provide any more details.

I would be happy to provide more logs if that would help.

g1t-dlanor, 2016-08-02:
Perhaps it is hard to describe every detail, but just knowing that it is while browsing videos tells us 100% more than what we knew before your last post, which was just that the GUI freezes for you 'in some unidentified context'. I assume that your last post meant that it is after entering a media section like "Movies" and browsing the list of posters in there that the freezing occurs. Right ?
Or is it simply when browsing video posters on the 'Home screen' ?

Additional details that we need to know include whether or not that freezing is just temporary, or completely fatal to the session. In other words: Do you have to pull the power plug and restart from scratch to regain control, or does the freezing let up after some delay ?

If it is just some temporary freezing delay, it might be related to the transfer of poster images from the server. Those transfers may then be eliminated by using the "Control Precaching" command in the side menu, so as to ensure that all poster images are cached in the RasPlex SD card.

On the other hand, looking at the log file you posted it contains lots of errors related to the transfer of poster images from your Plex server. Such messages could indicate a problem with the stored poster images on the server, in which case the precaching would probably also fail.
Have you tested that browsing these media sections works fine with some other Plex client ?
(Such as the Plex/Web client or the 'Plex for Android' client, for example.)
An ideal comparison test would be to use "OpenPHT v1.6.2.123" on another PC to test if that too has the same problem (as it is built from nearly identical source code as RasPlex).

Other suspicious error messages near the end of your log (presumably where the freezing occurred) show that something is wrong with some audio playback attempt. I'm not sure why it would be trying to play audio here, when you were browsing the media section, except possibly for keyclicks or perhaps show themes if your were browsing a "TV Shows" section with theme music activated in preferences.

As you can see from what I said above there are several details of interest that you could tell us about the circumstances where the freezing occurs, as well as some tests that could help clarify whether the problem is indeed caused by RasPlex or by some other circumstance. The possible suspects are many, including potential PMS database errors or SD card corruption. (The latter is likely if you ever had to 'pull the plug' to regain control.)

HappyCerberus, 2016-08-02:
The freeze is always permanent. So hard reset is required.

Other clients work just fine (web, android). I don't know how to locate the specific problematic videos that cause the caching errors, since I'm not really seeing any (visible) errors even on the Rasplex, just the freezes.

I have seen freezes on a second unit connected to different TV (same server), but that one is much less used, so I don't have any logs for that.

As for the location where the freezes occur. I'm really not able to pinpoint a specific location. They do occur most frequently sort of in-between episodes of TV show (when switching to the next episode), but that is also the most common action, so that may be the reason for higher freeze count.

The audio could be the theme music, which I do have enabled.

g1t-dlanor, 2016-08-02:
I see, this clarifies many things (though not solving the problem of course).
One thing I now need to ask is how do you make that "hard reset" ?

Pulling the plug or otherwise disconnecting power without proper shutdown causes SD corruption !!!
Only if the CPU is fully frozen (very rare) is it safe to do so (no ongoing SD card write operations).
But for all cases where the system is still producing log files (== SD card writing), corruption will occur.

A better way of restarting rasplex if the program has crashed but the LibreELEC system is still running is to connect to the RPi through an SSH client, logged in as user=root + pass=rasplex, and issue the command "killall plexhometheater". That will terminate RasPlex and restart it again.

But if you have 'pulled plug' in the past then I strongly recommend reflashing the card, installing and re-configuring RasPlex from scratch, as accumulated SD card corruption may now be the main cause of your problems.

closed 2023-02-25 by HappyCerberus