RasPlex/RasPlex #528: SSH not working on 1.6.2

doudz, 2016-07-30:
I'm trying to connect to my rasplex using ssh but it doesn't work, even if ssh service is enabled in Preference

**RPi model**: rpi zero
**RasPlex**: 1.6.2
**Network**: ethernet

I got this :
ssh: connect to host 192.168.0.21 port 22: Connection refused

g1t-dlanor, 2016-08-01:
What method did you use in your attempt to connect to the SSH server of RasPlex ?
Did you use the correct username="root" and password="rasplex" ?

doudz, 2016-08-01:
I'm trying to connect from gnome terminal on my ubuntu laptop
doudz@le_moyen:~$ ssh root@192.168.0.21
ssh: connect to host 192.168.0.21 port 22: Connection refused

ping works
doudz@le_moyen:~$ ping 192.168.0.21
PING 192.168.0.21 (192.168.0.21) 56(84) bytes of data.
64 bytes from 192.168.0.21: icmp_seq=1 ttl=64 time=1.39 ms

scanning using nmap show that port 22 is not opened so I guess ssh service is not started...
doudz@le_moyen:~$ nmap -T Aggressive -v -n 192.168.0.21

Starting Nmap 7.01 ( https://nmap.org ) at 2016-08-01 21:35 CEST
Initiating Ping Scan at 21:35
Scanning 192.168.0.21 [2 ports]
Completed Ping Scan at 21:35, 0.00s elapsed (1 total hosts)
Initiating Connect Scan at 21:35
Scanning 192.168.0.21 [1000 ports]
Discovered open port 445/tcp on 192.168.0.21
Discovered open port 3005/tcp on 192.168.0.21
Completed Connect Scan at 21:35, 0.15s elapsed (1000 total ports)
Nmap scan report for 192.168.0.21
Host is up (0.013s latency).
Not shown: 998 closed ports
PORT     STATE SERVICE
445/tcp  open  microsoft-ds
3005/tcp open  deslogin

Read data files from: /usr/bin/../share/nmap
Nmap done: 1 IP address (1 host up) scanned in 0.25 seconds

I tried to disable then enable ssh in Preference without any change

g1t-dlanor, 2016-08-02:
I have  no problems connecting to my own RPi1 or RPi2 devices using the same command format that you describe, in a terminal window opened on a system of mine running "Ubuntu-Mate 14.04".
For my  RPi2 case the command line used was:
ronald@RA6:~$root@192.168.10.147

On the first attempt this resulted in a system query  as to whether I really wanted to add this  address to the known host list, and after I verified this it proceeded to prompt me for the 'rasplex' password with the text:
root@192.168.10.147's password:

And after  that password entry I was connected to the RPi2 and could use SSH commands to list its contents etc. And this is how it should work for any normally networked RPi running RasPlex.
The SSH server of RasPlex should be active by default, listening for calls to port 22, though it is possible to disable it in the System settings menu (which I understand you already tried to use for toggling it off/on).

The only big difference between your failing setup and my working setup is that you're using an RPi0 without a native LAN interface, which points the finger at that USB Ethernet adapter as possibly interfering with the traffic. Have you tested that it really allows incoming SSH connections when connected to a normal computer ? That's something I'd recommend testing.

Even if that test does work, it could still be due to the USB adapter that the problem occurs, as the SSH server launch may need to find the network before that USB device has been initialized... :(
If so, then it will require changes to the LibreELEC code to change that behaviour.

doudz, 2016-08-04:
It's working now.
What I did :
- Try to find a log file by putting the microSD in my laptop : Ubuntu doesn't recognize the file system, on the second attempt it works but I suspect that the file system is a bit broken.
- Try to flash again the microSD with Rasplex using Rasplex installer : Rasplex refuse to boot saying file system is broken
- I suspect the microSD is faulty but I flash it again using ddrescue (not Rasplex installer) : it works !!!

So I think the problem was a badly copied Rasplex image on the microSD.

I'm closing the issue since it's not an issue.

closed 2016-08-04 by doudz