RasPlex/RasPlex #529: RasPlex shows cached screen of item not shared to managed user

WebSpider, 2016-08-05:
(referred to this issue tracker from the forum, https://forums.plex.tv/discussion/comment/1230746#Comment_1230746)

I have Rasplex / OpenPHT running on a RasPi 2, using the RasPlex 1.6.3 image, and have managed users set up on my account.

One of the managed users is for kids, which have a limited set of shared libraries (specifically: Only kids stuff, with an age-rating limit of NL6, TV-Y, TV-G in place)

Whenever someone else logs in with it's PIN protected profile, watches something else (say ... something TV-MA-rated) in a library not shared with the kids account, finishes, and turns off the TV, OpenPHT returns to the screensaver. No problem.

When then, the next morning, the kids log in to their account, the menu returns to the TV-MA rated item from yesterday, in a non-shared library with the kids account. Obviously, since the library is not shared with the kids, I didn't expect this to happen. I also can't find a switch to change this behaviour? It would be great if the menu position could switch back to default whenever a different user logs in..

**RPi model**:  2
**RasPlex**: 1.6.3 with default and Aeon skin
**PMS**: 1.0.3 on Linux Debian
**Network**: 100Mb wired ethernet

g1t-dlanor, 2016-08-06:
I agree. A new user login should always return the application to the top-level home menu, which should then show only the media section entries that are shared to the currently logged in user. 
Another potential problem of related kind is that both the background fanart slideshow and the screensaver fanart slideshow also need to be limited to fanart from the sections shared to the current user, so as to avoid showing children any shocking images. I'm not certain how this is implemented at present...

WebSpider, 2016-08-07:
The slideshow is a different (but related) issue. When I have some time and am not on mobile connection, I'll check if it already exists and if not create it.