RasPlex/RasPlex #532: suggesting an extra few seconds before "no response from network" error displays

checktravis, 2016-08-15:
When I first attempted the initial set up wizard, when it got to Networking, it took a while for the SSID list to populate (maybe 45 or 60 seconds). Then, after I put in my password it said there was a "imput/output" error. I googled and found that there is better results using a powered up I was just using a direct connection through an adapter to the mini usb on the zero). 

I tried again with a powered hub, and the SSID list populated instantaneously. This time, after I provided the password, RasPlex waited maybe 20 seconds before displaying that there was no response from the network. Less than a second after that message displayed the ip address populated on the screen behind it.

I am sorry I cannot supply a log since this happened during setup, before I could turn debugging.

I am simply **suggesting an extra few seconds before that error displays.** That was the only (knock on wood) issue so far, it's running better on the zero than it did on my Pi3. And, I can do everything through CEC, never needed a keyboard. On the pi, the CEC would cause it too lock up half the time.

Killer product here, I struggled for months working the kinks out of Plex on a pi3 and the o-c2 until I followed a suggestion to use RasPlex and separate the client from the server. Nice work :)

**RPi model**: pi zero
**RasPlex**: 1.6.2.123-e23a7eef; openpht skin
**PMS**: 1.0.3.2461  on armbian (odroid-c2)
**Network**: wifi N

thecorax, 2016-08-17:
(Sorry for Off-Topic.)
How is the performance on a Pi Zero as client?  The playback should be flawless for 1080p (~10GB files), right? But how is the menu? I am interested in a general 'look and feel' statement.
Thank you!

checktravis, 2016-08-17:
The install and setup was butter smooth. Biggest difference I noticed was how stable CEC is on the zero. With the 3, i had to ssh and restart it every other time I went to watch something.  It takes a bit longer to build the cache, and each time I start a movie it takes a few seconds to start. Once it was playing, no stuttering at 1080p.

It worked very well... until it didn't. I had to use a wifi dongle on a powered usb hub, and i that seemed to provide a bit of power to the zero as well as it's power usb port. I am assuming that is the issue because it just stopped working not long after I posted this issue here. I'm using the same (pihut) power adapter, but it seems the zero is less forgiving to fluctuations (i monitor the power with one of those USB meter things). In any case, the sd card was corrupted, so I re flashed the same card with the 2/3 version and I am back on the raspberry pi. It's slightly less smooth but overall considerably more stable so it's worth it  (in my opinion)