RasPlex/RasPlex #535: Skip backwards sometimes goes all the way back to the beginning of the video.

quinnmj, 2016-08-31:
Not sure whether this is a PMS issue or Rasplex issue - skipping back sometimes skips all the way back to the beginning of the video.  It happens occasionally after unpausing. I don't recall whether I've seen it happen at other times - I'll pay more attention.  

I'll see if I can duplicate it and get a debug log in the next day or 2.

**RPi model**: Pi 3 model B
**RasPlex**: latest stable 1.6.2
**PMS**: 6.1.9 running as Docker in Unraid
**Network**: 1GB Ethernet

benjaminwolf, 2017-04-11:
I believe this is fixed in the latest versions of PMS and OpenPHT/Rasplex 1.8.0.148.

g1t-dlanor, 2017-04-11:
One possible reason for unwanted 'skip to start' effects (even in current version) is if you press the skip buttons too fast, since the current implementation increases the skip distance when you do that, and a few very fast clicks can then go much further than the same number of clicks at a slower pace. (eg: after a few quickly repeated left-arrow clicks, each one can go like 15 minutes or more instead of the normal 15 seconds)