RasPlex/RasPlex #536: Feature request: Add mt7610u  drivers

Fluffinko, 2016-09-13:
Hi, is it posible to add mt7610u drivers in next release ?

RPi model: Rpi2
RasPlex: 1.6.2.123-e23a7eef
RasPlex git: 7580777a4bed0010d190ac3f04ebc686bf28a414
Network: [Wireless]

wizard872, 2016-10-04:
i want this driver too.

/lib/firmware 
mt7601u  is not working...

maybe need [ mt7610u.bin]  i think....

here link,  mediatek's  mt7610u-usb  linux driver.

http://cdn-cw.mediatek.com/Downloads/linux/mt7610u_wifi_sta_v3002_dpo_20130916.tar.bz2
http://www.mediatek.com/en/downloads1/downloads/mt7610u-usb/

dummys, 2017-02-10:
Same here, please add it or help us to install it. We can't install anything in rasplex...

WebSpider, 2017-02-10:
The module is not in the linux kernel,  and thus requires extra work every time a new release is made. Also check https://github.com/raspberrypi/linux/issues/730  there are some posts there with instructions how to do this as well

dummys, 2017-02-10:
Oh okay, So I think I should build my own version of Rasplex in order to add it ? Did you have some instruction on how to do it ? I'm actually generating the dev build environment using the developer guide of your github page.

dummys, 2017-02-10:
Yes it is now, since the 4.2 ...