RasPlex/RasPlex #538: Hangs on OpenPHT splash screen when starting

rdig, 2016-09-20:
When booting up the device (after the first restart where it resizes it's partitions) a splash screen appears showing the `Open PHT `logo, then it just hangs, and won't proceed further.

I can see it in the list of players in the plex web interface, but it fails to connect to it.

**Steps i've tried:** 
- I've written the microSD using the RasPlex linux installer, and manually using `dd`
- ~~Left it running for 2 hours and still the same (though it was building caches / doing initial setup)~~
- Left it running overnight now, still the same.
- Tried two previous image versions (`1.6.1` and `1.6.0`)
- Tried two different cards, 8 Gb and 16 Gb, both Class 10

I'm beginning to suspect it might be a problem with the Raspberry Pi.

*Side note*: tried installing and running `Kodi` via `LibreElec` and it works just fine.

**RPi model**: Raspberry Pi 1 model B+
**RasPlex**: RasPlex-1.6.2.123-e23a7eef (don't know what skin, guessing the default one)
**PMS**: Version 1.1.4.2757 on Ubuntu 16.04
**Network**: Ethernet

CoalaJoe, 2017-05-03:
Same problem here... A shame this project is dead.

rdig, 2017-05-03:
> A shame this project is dead

Maybe it has problems on older models of `Pi`s, but that's an outlandish claim. 

I'm happily using this on a `Raspberry Pi 3` w/o a hitch for almost 6 months now.

And given that the last commit _(f53d62d)_  was on Mar 19, this is most likely not dead.

CoalaJoe, 2017-05-04:
Ok. So with a Pi 1 im out of luck..

g1t-dlanor, 2017-05-04:
@CoalaJoe 
That depends on what model of RPi1 you have. The old RPi1a model only has 256MB of RAM, which is insufficient to run RasPlex OpenPHT properly. But the RPi1b model normally has 512MB of RAM (with possible exception for the very first batch of them), and those work fine with most versions of RasPlex, including the current release v1.8.0.148. As a beta tester for the project I always run the latest version both on an RPi1b and on an RPi2b, and the current version runs fine on both, though obviously with some speed limitations on the single-core RPi1b as compared to the quad-core RPi2b. In addition two of my relatives also use RasPlex on RPi1 to access my PMS server remotely, without any problems so far. So I'm fairly sure that the problem is due to something specific to your hardware or its configuration.

As for giving any detailed help with your problem, that's nearly impossible with such a vague description of it as given in your first post. A simple "same problem here" doesn't necessarily mean that you're using identical setup and configuration as the original poster. The only tip I can give you is that 'hanging on splash during boot' problems in nearly all cases are due to either insufficient PSU or low-quality SD cards. (Some claims to class 10 are fake...)

CoalaJoe, 2017-05-04:
@g1t-dlanor I got a really old Pi 1. It doesn't even have mounting holes. The PSU is no problem, I've powered other Pis with it. Aswell as the SD-Card.

"A shame this project is dead."
This issue is more than a half year old and noone answered him, or stated that they will look into.

dalehamel, 2017-05-04:
@CoalaJoe claiming that the project is dead doesn't add to a productive conversation.

> This issue is more than a half year old and noone answered him, or stated that they will look into.

Everyone working on this project is doing so on a volunteer basis, and without enough information to reproduce a problem (which is rarely given, as it can be difficult to do so) it can be hard to look into problems.

If you are able to SSH in and post logs, we might be able to see why there is an issue loading. If you're unable to ssh in, there are still ways to obtain logs off of the SD card.

I don't believe we enable swap by default, doing so should make up for the lack of memory on the original Pi board, which I believe some other distributions do. If I recally, this can be done with a setting in the config file, but I don't remember where.

CoalaJoe, 2017-05-04:
@dalehamel Thanks. I will look into it.

CoalaJoe, 2017-05-07:
I tried the following:
1. `chroot` into Raspberry's SD-card. (Couldn't quiet figure that one out. Maybe because of architecture difference or Libreelec)
2. Tried to connect via SSH. Connection was refused, probably didn't initialize OpenSSH-Server.

A port scan does not show any open ports.

scorpuk, 2017-05-07:
Same problem here.

Also connection refused when trying to SSH with putty.

RPi model: Raspberry Pi 1 model B
RasPlex: RasPlex-1.8.0.148-573b6d73-RPi.arm.img
PMS: Version 1.5.5.3634  on Ubuntu 16.04
Network: Ethernet