RasPlex/RasPlex #539: Feature request : autoplay next episode of a season

samshoo, 2016-10-03:
Hello,

I'd like to request the **auto-play next episode of a season** feature ! 

Where the next episode would start automatically when an episode ends. The feature should be available whether the episode was launched from a TV Series page or from "On-desk". 

We should be able to set the feature on or off  and the timer (time before starting the next episode in seconds) in the settings.

Let the binge-watching begin !

S.

bkury, 2016-10-03:
Isn't that feature already available?

On tvshow level --> select show --> context menu --> play (will auto-play the complete tvshow)
On season level --> select season --> context menu --> play (will auto-play all episodes for this season)
On episode level --> select episode --> context menu --> play from here (will auto-play all remaining episodes)

Isn't that what you want?

samshoo, 2016-10-04:
Seems to be exactly what I want, I wasn't aware of this, because my remote doesn't allow me to open the context menu...

closed 2016-10-04 by bkury