RasPlex/RasPlex #540: Direct Play will not work

jpriddy, 2016-10-13:
I have a RasPi2 running rasplex 1.6.2.123-e23a7eef . I have modified config.txt to enable the 'Pi2' overclocking settings. I have changed my settings to enable passthrough to my home theater (which supports DTS, etc and had 5.1 speakers attached). I have precahed my library items. But, when I try to direct play much (but not all) of my media it fails with 'Unable to play media.' I don't know that a true pattern has emerged, but it seems like the 'higher quality' media I have is typically the stuff that fails.  If I change my plex network settings for local to 'maximum' from 'direct play' **OR** if I check the 'force transcode' these exact same media items that failed before, now play fine (although it taxes my plex server needlessly).

Am I missing something? Is RasPlex not designed to direct play items due to hardware limitations, etc? I see some people posting about buying codecs and such -- something I have never had to do with PMS or OPHT on mac or windows.

Logs from rasplex: http://pastebin.com/QPYZdr01

Logs from pms: http://pastebin.com/gtGF1Jvu

Media comparison: http://pastebin.com/vWTjrH8W

pi:~ # tvservice -a
PCM supported: Max channels: 8, Max samplerate: 192kHz, Max samplesize 24 bits.
AC3 supported: Max channels: 6, Max samplerate:  48kHz, Max rate  640 kb/s.
DTS supported: Max channels: 6, Max samplerate:  96kHz, Max rate 1536 kb/s.
EAC3 supported: Max channels: 8, Max samplerate:  48kHz, Max rate    8 kb/s.
DTS_HD supported: Max channels: 8, Max samplerate: 192kHz, Max rate    8 kb/s.
MLP supported: Max channels: 8, Max samplerate: 192kHz, Max rate    8 kb/s.

**RPi model**: RasPi2
**RasPlex**: 1.6.2.123-e23a7eef 
**PMS**: Linux (RHEL7.1+ all updates) w/ Plex Server 1.2.2.2857
**Network**: Ethernet 100mb

Kwiboo, 2016-10-13:
This issue is caused by your server bandwidth setting (20mbit result in a 16000 kbps limit) and the new Streaming Brain server feature, see `Streaming Resource: Bandwidth exceeded: 20690 kbps > 16000 kbps` in your PMS log.

A fix to make OpenPHT/RasPlex more Streaming Brain compliant is in the works, but that fix would only have made RasPlex trigger transcoding due to your bandwidth limit instead of failing when trying to direct-play a remote file.

Are your server on same network or is it a remote server?

jpriddy, 2016-10-13:
Thanks for the quick reply.  

Everything is local -- same switch as my PMS.

Pardon my ignorance, but if I am reading this correctly,  the hardware itself just can't handle a direct stream for some of my media that is encoded at a larger bitrates and the media needs to be 'stepped down' via transcoding on the PMS side beforehand for rasplex to be able to handle it?

So is that to say that very high quality streams will never be able to direct play with rasplex on this hardware?  Would the pi3 fare any better?

Kwiboo, 2016-10-13:
The hardware should be capable, the VideoCore IV GPU can usually manage 1080p h265 level 4.1 8 bit depth video streams up to 40Mbit without any problem.

I believe the issue is that your server for some reason think your local client is a remote client and enforces a bandwidth limit that restricts the rasplex client from direct-playing movies with a bitrate over 16000 kbps.
If you have set a `Internet upload speed` or `Limit remote stream quality` under `Settings > Server > Remote Access` I suggest you increase these values and see if it helps.
Full log-files will also be very helpful and give us a full picture of what is going on.

jpriddy, 2016-10-13:
Indeed -- I had it set to 20mb, though as you said it shouldn't come into play since its local.  Regardless I will remove that setting and bump up the verbosity on both logs.  Stay tuned.

jpriddy, 2016-10-13:
Disabling the remote access network throttling resolved my direct play issues so far (only tried with a handful of items but 100% success so far).   If you want the logs for either the client or server to troubleshoot why it thinks its remote instead of local let me know and I'll kick out everyone streaming so I can get some clean logs for you.

Kwiboo, 2016-10-13:
Great that it worked out but very strange that it gets treated as a remote client. I am currently working on improving Streaming Brain support to resolve this type of issues.

Logs from the client would be helpful in understanding why it is treated as a remote client, please enable debug logging in RasPlex under `Preferences > System > Advanced` and try to capture a clean client log when you have the 20mbit throttling set on server and try to play a video that fails.

jpriddy, 2016-10-14:
Failing to play content 2x with 15mb throttling set on server:  http://pastebin.com/vg1bBz0X

Successful play **without** any throttling set on server:  http://pastebin.com/3fPxu3G3

BTW thanks for your help, it probably would have taken me a while to dissect the logs and try disabling the throttling server side -- you saved me a boatload of time and hassle.  So far so good with pretty much every other piece of media I have tried since I made that change.  

Let me know if you need anything else -- more than happy to help out.  Beyond that feel free to close out this issue.

andruare, 2017-02-26:
Still having the same issue with the latest version 1.7.1 on rPi 2.

Is this issue is still open? Is there a workaround? :)

g1t-dlanor, 2017-02-26:
I doubt very much that it can be the same issue as the one @jpriddy had, since he found a working solution which you should also apply if you truly have the same issue. Also, many others (myself included) use RasPlex on RPi2 daily with no problems at all with direct-play playback, so there must be something specific to your setup or usage that causes the problems you have.

And as of yet we know no details about your problem except that it should have something to do with Direct Play not working (since that is in the issue title), and that it should NOT be identical to the problem of the original poster, since his cure for the problem didn't work for you (or simply wasn't applied).

Please describe your problem in greater detail, as well as any measures you may have tried to fix it, with or without any change in the symptoms. Then we can try to advice you on how to get better results.

andruare, 2017-02-27:
Sorry for the short message...

Yes ,the solution is to switch off the remote bandwidth restriction. Then Direct Play works fine.

But with the activated restriction for remote servers (!) direct play is not possible because Rasplay identified the server as a remote one. The Plex Media Server is a local one on the same subnet ipv4.

Is it possible to force Rasplex to identify the server as a local one?

g1t-dlanor, 2017-02-28:
I'm not entirely sure how the Plex client and the PMS server negotiate the connection and decide whether it is remote or local. Thus I'm not sure whether that decision is ultimately made by the client or by the server. But I do know that one of the complications involved is due to the PMS server reporting all existing IP addresses of the server PC as belonging to the PMS server and to be used for accessing it (even internal/virtual IPs that are never intended for PMS use). And since such IP addresses need not be part of the same IP range as the main LAN, they can confuse the software into treating a local connection as remote.

One  thing you can try is to use the RasPlex submenu: "Preferences" > "Media Server"
There you can define a specific IP address to be used for the server, and when that IP address is in the same LAN IP range as the RPi itself, the connection should definitely be considered as local.

Please let us know if that method works for you.

andruare, 2017-02-28:
YES, this works :)
Manual server IP and direct play works with activated bandwidth restrictions :)

Thank you so much! <3