RasPlex/RasPlex #542: Run script on play/pause/stop

sticilface, 2016-11-01:
Is it possible to run a script on play/pause.  

I'm wanting to integrate with home automation to switch TV on, select correct channel etc.

Is this possible?  

Thanks for all the good work.  rasplex is the centre of my media watching, used every day!

jishi, 2016-11-28:
Coming from another angle at this problem, I'm using the JSON-RPC support that this player has inherited from XBMC/Kodi. By default, it only listens to connections from localhost, but it's possible to switch if you modify one of the setting files in your installation.

This means that I have an application (written in node.js, but any language that can do socket connections would suffice), using this npm module (that I have patched to correctly identify dead sockets) https://github.com/jishi/xbmc-listener.

I use this to control my Sonos system and switch to line-in automatically when plex starts playing, as well as pauses all other players. 

If you are interested, I can share my simple script.

sticilface, 2017-02-11:
I just found this thread whilst searching for a way to do it.   Thanks for the offer, but i'd like to keep it as simple as... so preferably have it integrated into rasplex.  I'm sure there would be a number of people who would use it, for exactly this reason.  I'd like to switch on hue, turn the stereo on, and switch the amp channel... when content is played.

jishi, 2017-02-11:
Well, I find it unlikely that they will give you exactly what you want when it already has the JSON-rpc functionality. 

I recently bought the Philips hue as well, and have now added scene switching when I start playback, which will dim down the lights when I start playback

sticilface, 2017-02-12:
I got it working. I failed to read your first reply carefully. Thanks.

closed 2017-02-12 by sticilface