RasPlex/RasPlex #545: Missing list by folders option in RasPlex interface

serenity-networks, 2016-11-04:
Rasplex is missing a standard Plex feature in it's interface.  When selecting a "movies" type folder, Rasplex is missing the List by Folder sorting option.  This is available on Roku, PC, iOS, and Android interfaces.  It happens to be one I depend on at my daughters school for their robotics team and raspberry pi after school clubs I help with.  All of there instructional videos are on plex, and sorted by folder.  Plex is not capable of organizing these on it's own, since many of them were home made, or taken from you (youtube is blocked at the school).  We have a plex plugin that will snag the descriptions from youtube and add it into the meta data for those, the rest I manually put in, and I manually organize the folders.  I REALLY want to deploy a few more raspberry pi's running rasplex at the school for the students so they can watch the video's, but having this functionality is key.  There are over 1000 video's and we must retain the folder structure through the interface.  Can this please be addressed and made available?  It seems like it is most likely a skin issue.  The data is obviously provided to the plex clients because all the other plex widgets can get at it.

**RPi model**: [All Models]
**RasPlex**: [Default skin, and Plex-dark]
**PMS**: [1.6.2.123]
**Network**: [ethernet and wifi]

Thanks for all the work you guys put into this!

WebSpider, 2016-11-04:
I think this is present in the Aeon-Nox theme. Have you tried that out?

serenity-networks, 2016-11-04:
I have not tried that.  Only the default Plex theme, and Plex-dark.  I will try this today and post the results here.  Thank you for the information!

serenity-networks, 2016-11-04:
I've installed Aeon-Nox.  I have looked extensively in the same views area, and although the skin is awesome and provides a lot of customization, I cannot seem to find any way to show or list by folder.  Do you know something I'm missing here?

serenity-networks, 2016-11-04:
It's also worth noting, Aeon-Nox, on a fresh RasPlex install, completely crashed OpenPHT.  Obviously, if there is a way to display folders, I would put the work in to make it work somehow.  Thanks

bkury, 2016-11-04:
For now you could add your educational movies as "Home Videos" (PMS) and you should be able to use the "By Folder" filter in OpenPHT

I'll try to find a better solution ...

serenity-networks, 2016-11-04:
Yes, that could be a solution, but we loose all of the descriptions for each video, so the kids have no idea what the video is, other than the name (nor do I or the other teachers, staff and volunteers).  Some videos have a lot of topics, which is covered in the description.  The descriptions also have library codes that correspond to documentation the kids have for projects.  I.E. how to build the circuit board for the widget in _this_ video, or what virtual machine image to use to reproduce the network scanning that was shown in _this other_ video.  (in simple terms).  We have thousands of these that have been accumulated.  

Since we've decided to try to implement RasPlex, We've found a youtube description plugin that looks for the video id in the file name (thank god we incorporated that into the last 11 characters of each file name using youtube-dl) that pulls down the youtube descriptions (this accounts for about 25% of the library), and another plugin that someone kludged the crap out of that will read meta data from a text file that has the same name as the video file, when we can actually get it to work (for all of the non youtube stuff, sometimes we have to remove and add a library 5 or 6 times before it will parse the text files, total PITA).  If we set our library types as home videos, plex won't allow us to use any "movie" agent plugins, which unfortunately is what the authors of those plugins configured them to be.

One thing that would be incredibly useful to the plex community, in general, is a plugin that would read metadata (at least a description) from a .txt/.summary/.meta/.whatever file with the same name as the video and insert that into the plex db.  And, said plugin or agent work with any library type, especially the home video type as that's they only type that doesn't try to rename and parse online databases to screw up naming conventions.  I've talked to a lot of people in the plex community on my journey to use rasplex at the school and they all seem to have a weird reason to need this.

But a good start point for my particular situation, and in general, would be allowing use to view by folder in movie libraries.

Thank you EVERYONE for your input and attention to this.  I'm usually completely on my own when it comes to the electronics, robotics and Pi clubs at the school.  It's nice to know there is some help out there.

bkury, 2016-11-05:
I see, another workaround you could try is to remove/rename the file "DialogFilters.xml" in your installation directory (C:\[INSTALLDIR]\OpenPHT\addons\skin.plex\720p)

You should now get a list of available filters (including by folder) as soon as you enter a media section.

bmemoret, 2018-11-17:
I completely agree that folders are the most useful classification of movies, because it's personalized.  In addition, metadata gathered about a movie from the web results is gross misclassification a lot of the time (such as horror movies put in family movies, or James Bond movies ending under romance, etc.)
Even when some folders might coincide with the automatic classification (e.g., by director), bad metadata is so prevalent that half of the movies directed by the chosen director don't show up.
It's much faster putting files in suitable folders than editing the metadata by hand.
So, please, whatever can be done to restore browsing by folders would be great!
(I cannot use KODI, which has a "by folder" feature, because of legal troubles between ASUSTOR -- my NAS brand -- and some unnamed organization forbidding them to support AC3/DTS playback.)