RasPlex/RasPlex #546: Error playing file when server has bandwidth limit

splu0008, 2016-11-09:
I noticed an error on my Raspberry Pi with RasPlex installed. When I enable a bandwidth limiter on the PMS it gives the error: Failed to failed to play, item not found. When I disable the bandwidth limiter it works again. 

Can you please fix this?

**RPi model**: 1B
**RasPlex**: v1.6.2.123 default skin
**PMS**: Linux 1.2.7.2987
**Network**: Ethernet

g1t-dlanor, 2016-11-10:
If I understand your problem correctly then it's not due to the coding of RasPlex but is rather a problem with the PMS implementation of bandwidth limiting. AFAIK the client program (here RasPlex) is not aware of server limitation settings, so it can't compensate for them. It's the responsibility of the PMS code to recognize when a media playback request requires bandwidth beyond the set server limit and then transcode the media to a lower bitrate, instead of just flatly refusing to deliver the media like you describe.

The client is only responsible for transcoding decisions when transcoding is needed due to settings of the client, such as its own bandwidth limit settings or CODEC compatibility issues. But you probably could make RasPlex request transcoding by setting its bitrate limit to something slightly lower than the limit you set for PMS. (Just a workaround, not a real solution...)

If I missed some aspect of your problem that would make it a client issue, feel free to clarify this in a future post here. But I really think that you need to report this issue to the PMS developers.

splu0008, 2016-11-16:
g1t-dlanor, thanks for the response, I 'fixed' it by enabling the setting: Force Transcoding.