RasPlex/RasPlex #549: Unlinked Plex after restart

Eloo, 2016-12-02:
Hey Guys, 
currently I'm facing an issue with The Connection/linking between a RasPlex instance an my PMS Server (Remote).
If I Link my RasPlex instance via Plex/Pin or plain credentials the settings seems to disappear after a reboot. More detailed, the "sign in" option reappears in the submenu but my libraries are still there. The account information in the settings stucks at "trying".
More critial is the fact, that also the libraries seem to disappear after a random time when the RPI is shutdown or disconnected from the power supply and I need to "relink" this rasplex instance.

Anyone facing the same problem? Or better, anyone having a possible solution or workaround?

**RPi model**: RPI 2
**RasPlex**: 1.6.2, 1.7.1 with Plex Black Edition
**PMS**: 1.2.7 (latest before 1.3.0)
**Network**: Ethernet & Wifi

Thanks for your great work.
Best regards
Eloo

ryantg3611, 2017-01-20:
I have the exact same issue!! I paid for a plex account on eBay. The seller has to sign me via pin. After doing that it works really well! All the Plex clients i've tried didn't merge all the servers into one. With Rasplex i can just search for the Movie/ TV show and thats it!! 

For the past week i have been trying to keep it signed in but after a reboot or two it signs out and asks me to re-calibre Rasplex!!?? Not sure why this keeps messing up.

Eloo, 2017-02-15:
any updates here?

attzonko, 2017-09-04:
I am seeing the same issue with 1.8 

If I select restart from the main menu it restarts and I have to go through the account linking process all over again.

Eloo, 2017-09-05:
the issue is gone on my side and my best guess is an instable wifi connection.
after i changed my wifi setup it seems to work without any issues

closed 2022-10-19 by eloo