RasPlex/RasPlex #550: Newer version of lcdproc

MangaValk, 2016-12-13:
Can a newer version of lcdproc be added to RasPlex?
The current version of lcdproc in RasPlex is from may 2014 from what i can see.

The problem this version has that it cant show the progress bar or special icons except heartbeat.

For reference this person had the same issue already in January 2015, assuming that with pht he means RasPlex. https://forums.plex.tv/discussion/141945/pht-ubuntu-lcdproc-no-german-umlaute-and-no-progressbar

I would like to contribute and do this change myself too, but i don't understand how rasplex pulls the lcdproc version.

RPi model: RPI 2
RasPlex: 1.7.1
PMS: 1.3.2.3112
Network: Ethernet

MangaValk, 2017-04-26:
it seems the development of rasplex is abandoned.
its a shame :(

Kwiboo, 2017-04-26:
Please check https://forums.plex.tv/discussion/comment/1407549, the newest lcdproc addon should have fixed the issue reported in that thread.