RasPlex/RasPlex #552: Doesn't support the lastest Synology Plex Media Server(version: 1.3.3.3148)

goxofy, 2016-12-20:
[Tell us about your issue, steps to reproduce, expected and actual behavior]
[Always attempt to reproduce your issue in the default skin]
[Enable debug logging in preferences and insert a link to your log-file]

**RPi model**: [raspberry pi model]
**RasPlex**: [version and skin]
**PMS**: [version and operating system]
**Network**: [ethernet/wifi]

g1t-dlanor, 2016-12-20:
If you've recently upgraded from that older PMS version to the current one, then I urge you to doublecheck all the server settings, as some features have been added/modified in newer versions and the new default settings may not be appropriate for your setup.

Also, I'm not entirely sure of what you mean by "can't get any media info". It could mean that RasPlex gets no connection at all to your PMS, but it could also mean that it does connect and get a full media listing, just lacking all detailed media info, or anything in between these variants. I suspect the first variant is more true, but please describe your problem in greater detail so we'll know for sure.

Another thing we'd like to know is if you've successfully used any other Plex clients (on PC/Android/iOS) with the new PMS version, so we can rule out any problems with the functionality of that server as such.

goxofy, 2016-12-20:
Hello, dlanor

Thank you very much for reply.

I have checked the settings in the newest PMS and there is nothing different from the older one.

And I'm not sure whether the RasPlex gets connection to my PMS, I have specified the PMS ip:port in the RasPlex settings but the RasPlex can't get anything from the PMS, no media listing. no details. I suspect that  the RasPlex doesn't get connection to my PMS.

And the third thing, I have successfully used many clients with my PMS, including iOS, Web and Android(on the Xiaomi TV Box), so I think my PMS is ok.

Thank you very much!

Tink

Kwiboo, 2016-12-20:
@goxofy when using newer PMS versions you will need to sign-in to RasPlex (same as any other Plex client) or add your local network to the `List of IP addresses and networks that are allowed without auth` server setting

goxofy, 2016-12-21:
@Kwiboo Thank you!

Can you tell me how I can add local network to the "List of IP addresses and networks that are allowed without auth" server setting, I cant find this setting on my PMS.

goxofy, 2016-12-21:
@Kwiboo Thank you very much!

It works!

closed 2016-12-23 by goxofy