RasPlex/RasPlex #561: Alexa does not discover rasplex.

sticilface, 2017-02-03:
Alexa discovers my safari web player but not rasplex. 

any ideas why.  rasplex works fine with my browser and phone app... 

[Tell us about your issue, steps to reproduce, expected and actual behavior]
[Always attempt to reproduce your issue in the default skin]
[Enable debug logging in preferences and insert a link to your log-file]

**RPi model**: pi 2
**RasPlex**: 1.7.1.137-b604995c
**PMS**: Version 1.3.4.3285
**Network**: wired

tukten, 2017-02-03:
Same issue here. I know PHT isn't officially supported but given that casting works totally fine to RasPlex from all other plex clients I would've thought Alexa wouldn't have any issues.  Does the product attribute have to be something other than "plexhometheater" since PHT isn't officially supported anymore? 
Is there a way to make RasPlex advertise itself as PMP instead of PHT or just "Plex Web" perhaps?

lizaoreo, 2017-02-04:
https://forums.plex.tv/discussion/256779/alexa-for-rasplex

Based on that it looks like they need to incorporate Plex Companion support.

bitcraze1, 2017-02-12:
Any update on this? Very excited to have this working.

muyiwao221, 2017-02-12:
I as well am curious to any updates. I bought a raspberry  pi just for this capability and I realized last night that rasplex wasn't compatible. Definitely was a huge disappointment.

Kwiboo, 2017-02-12:
Support for Alexa is scheduled for next RasPlex/OpenPHT release, I have already made a small successful test and was able to receive the control commands from Alexa inside OpenPHT.

sticilface, 2017-02-26:
Let me know if you need someone to test it out?

1liminal1, 2017-03-02:
Any updates on this?

Thanks

jpriddy, 2017-03-23:
Latest version has support for it:  https://github.com/RasPlex/OpenPHT/releases

closed 2017-03-23 by dan-the-man-nz

bitcraze1, 2017-03-24:
Hi, RasPlex Support

Much, much, much appreciated!

sticilface, 2017-03-24:
This is great. thank you.

sticilface, 2017-03-28:
Alexa seems to loose the rasplex player every now and then.  A reboot seems to sort it, but it is around once a day.  again my other plex players, ie. web and iOS client discover it fine.  

Is there any debugging I can do?

winers1290, 2017-04-04:
I've downloaded the pre-release v 1.8 and successfully integrated Alexa with my Plex server, but PHT is not discoverable as a player. 

I've rebooted everything several times after the update. Should I try anything else?

tukten, 2017-04-04:
I had that happen too. I unlinked the Plex skill and then reenabled it, and it worked.

winers1290, 2017-04-04:
I have also tried this several times... It is categorically denying there is a player connected to my network.

g1t-dlanor, 2017-04-06:
@winers1290 
Is your RasPlex instance logged in to the PMS server via password or plex.tv login code ?
That may be a factor in how Alexa finds available Plex players.

RasPlex itself can work fine without being logged in that way, as can the PMS server itself.
(Depends on settings...)
But for some other services to interact with PMS servers and Plex clients those may need to be logged in.

winers1290, 2017-04-06:
It is logged in via plex.tv code. I'm unclear if you were saying this is the correct way, or should I try accessing the server directly?

g1t-dlanor, 2017-04-08:
Having both client and server logged in via plex.tv (whether by code or password) should allow the most functionality. So if you're already doing this, there's no point in changing it.
And if both are fully connected then I have no idea why Alexa can't find RasPlex.
Hopefully someone else will be able to find the solution.

winers1290, 2017-04-08:
I'll have another run going through and resetting/restarting everything. If that doesn't work I will create a separate issue, since this could be a problem in the latest update.

winers1290, 2017-04-08:
Figured this out.

1. I signed out of plex.tv and back in.

2. My RasPlex was signed in to plex.tv, but logged in as a managed user. Changing users to my plex.tv was step one. When I change back to the managed user, it still works, but it obviously helps to form the initial connection.