RasPlex/RasPlex #563: HDMI-CEC Quirks: Stuck in menus.

svin83, 2017-03-03:
If you use HDMI-CEC to control RasPlex, there's no way to exit the preferences menu. The "Back" or "Exit" buttons on the remote won't work with OpenELEC or Kodi either... This menu could have an exit button as a workaround.

**RPi model**: All
**RasPlex**: All
**PMS**: All
**Network**: All

chadberg, 2017-12-23:
This is likely to have more to do with your remote. If your remote does not have a 'return' or 'back' button you may find yourself in this circumstance. My tv remote includes a return button which works fine to return from these menus.

svin83, 2017-12-23:
Okay...

It does... It has a back/return button, an exit button, and a cancel
button...

closed 2020-09-01 by svein83