RasPlex/RasPlex #564: Does not work on Pi Zero W

2E0PGS, 2017-03-05:
Rasplex don't work on the new Pi Zero W.

Either shows nothing or shows rainbow screen. No activity LED either.

Tried it on Rasplex 1.6.2 and 1.7.1 to no avail.

**RPi model**: [Pi Zero W]
**RasPlex**: [1.7.1]
**PMS**: [Latest]
**Network**: [on-board WIFI same chip set as Pi 3 Model B]

dalehamel, 2017-03-05:
I've never been able to get my hands on a Pi zero to attempt to test it

falkyre, 2017-03-13:
I just ordered two Pi Zero W's and they are shipping, should receive them this week.  I can donate one of them to the cause.

ghighi34, 2017-03-13:
Hello guys, 

I've had the same issue with my brand new PI0W.

I've followed this tutorial and it's now working fine : 

https://www.youtube.com/watch?v=rvKHefzOTVc

Cheers,

kgorskowski, 2017-03-13:
@ghighi34 thanks for the link. can you tell with which version of RasPlex it worked for you? Thanks in advance!

dalehamel, 2017-03-13:
I have ordered some test units to ensure it works out of the box.

Adafruit finally has stock :)

bigmal, 2017-03-15:
@ghighi34 In GetRasPlex what model Pi and what version of Rasplex did you select?

ghighi34, 2017-03-15:
@bigmal @kgorskowski,

It's working with the following version : 1.7.1.137-b604995c

I hope it will help you.

benjaminwolf, 2017-03-25:
This working now as of 1.8.0.148

closed 2017-03-25 by benjaminwolf

varunmehta, 2017-04-28:
How is the performance on Pi 0, compared to Pi 2 ?

g1t-dlanor, 2017-05-01:
RPi0 has appx the same performance as the RPi1 since they both have single-core CPUs, which is the major difference between those models and the more advanced RPi2 and RPi3 that both have quad-core CPUs.