RasPlex/RasPlex #568: Rasplex wont allow me to select individual IPTV channels

jay2k6ie, 2017-03-22:
I have a weird issue with Rasplex and thought you might be able to help. I'm delighted with it but I cannot select individual IPTV channels when I select them. I highlight the channel I want to play and press enter (hear the noise for key press) but nothing happens. I've tried a mouse and clicking on the channel doesn't work either.. It basically just won't allow me to select a channel.

I can go to the play icon in the top left corner and play the entire folder of channels (which starts from the first channel in the folder) and then once playing press channel up to find the channel I want but its very tedious.

1st gen RBP model B
Rasplex 1.7.1

Wifi

dan-the-man-nz, 2017-03-22:
Please test on the latest version of Rasplex.

Also, what IPTV plugin is this?

What skin are you using?

jay2k6ie, 2017-03-22:
Is 1.7.1 not latest version? skin is openPHT. Not sure what the plugin is called its a paid service

dan-the-man-nz, 2017-03-22:
1.8.0 is the latest. the Cigaras IPTV v2.0.x generic IPTV plugin works perfectly in Rasplex/OpenPHT. this is probably  a issue with whatever channel you are using. you could also try using the Aeon Nox skin as i use that with IPTV to see if that makes any difference

There is not enough information has been provided to go any further with this issue as you cant tell us what plugin are you using etc, so im going to close this issue.

closed 2017-03-22 by dan-the-man-nz