RasPlex/RasPlex #570: WIFI not found due to channel

udochrist, 2017-03-25:
Hi,

 i've had this issue with basically all PI distributions. when it comes to WIFI setup there is no possibility to select the "region" to enable additional regulated bands. I'm located in germany and my AP choses band 13 (because it can and it is least congested).
Due to this i cannot select my WIFI when configuring RasPlex. and hence RasPlex is not usable at all.

Can you please enable a Country/Location selector loading the regulatory domains (only jessie images there are two to three additional packages needed) so channel 13 and others are active?

otherwise rasplex (at least the config screens i was able to see) looks amazing. Thanks for your work.
best regards
udo

**RPi model**: RaspBerry 3b
**RasPlex**: 1.8.0 (fresh install)
**PMS**: latest
**Network**: wifi (location is Germany, Channel is 13)

benjaminwolf, 2017-03-25:
Channel 13 is prohibited in some countries, so the wifi world code is enabled by default.

SSH in and try these commands:

mount /flash -o remount,rw
echo "options cfg80211 ieee80211_regdom=DE" > /etc/modprobe.d/cfg80211.conf
reboot

Replace 'DE' with your 2 character country code in the second line.

udochrist, 2017-03-25:
Hi

I know about regulatory domains. Fwiw the ssh thing is not something I would expect an end user to do. I also do not ssh into the box to set the timezone editing config files.

So: i think instead of the commands the fix should really be a configuration item in the network setup asking the user about the location/country.

Cheers

benjaminwolf, 2017-03-25:
I'll reopen the issue as you're requesting an enhancement to the GUI.