RasPlex/RasPlex #572: 1.8.0 not playing movies with external subtitles

edjonesy, 2017-04-07:
I updated RasPlex from v1.7.x to v1.8.0 yesterday, tried a few shows and films and they played fine. During the day yesterday, while I was at work, my wife tried to play a film which had an external .srt file for the English subtitles, she got an error message on RasPlex saying 'Unable to Play Media', I asked her to try other films, which all played, so I asked her to try another film with an external subtitles file, same error message. In all she tried 6 other films with external subtitles files, none of which would play, although I had watched some of them previously.

I logged in to my PMS server and was able to play the films in the web browser with no trouble.

I downloaded PLEX to my PS4, logged her in and she watched the film with external subtitles via the PS4 with no problem.

When I got home I tried to play several other files and every movie with external subtitles failed, movies without subtitles played fine.

I reinstalled v1.7.x onto the micro sd card and the files now play perfectly via RasPlex.

I don't have the logs, the wife was involved so everything becomes damage limitation, make it work now if not sooner...

All movies/TV shows are played direct, I transcode them prior to adding to my library with Handbrake, the movies etc. are all .mp4 format, where external subtitles are required their naming convention is 'title.en.forced.srt' to force PLEX to serve the English subtitles, they are a mixture of 480p, 720p and the odd 1080p.

Sorry to have found a bug, hope you can replicate it and fix it.

**RPi model**:  Raspberry Pi Model 2
**RasPlex**: 1.8.0 default skin
**PMS**: PMS v0.9.16.6 / 2006 Mac Pro running OSX 10.7.5
**Network**: PMS Gigabit Ethernet / RasPlex Wireless N WiFi

benjaminwolf, 2017-04-11:
I cannot replicate this issue with PMS Version 1.5.3.3580 on Windows.

edjonesy, 2017-04-11:
Perhaps it's something to do with my particular setup, my server won't run PMS later than v0.9.16.6.

I'll install RasPlex v1.8 onto another Micro SD card, and enable logs, I'll see what happens then.

Is it better to set RasPlex to the Pre-Release channel and allow it to update or to install from scratch using the v1.8 image (RasPlex-1.8.0.148-573b6d73-RPi2.arm.img.gz)?

Thanks

g1t-dlanor, 2017-04-11:
@edjonesy 
For a new install of 1.8.0 I would use whatever method you did not use previously, since that one left this issue. So if you used the internal update method last time, then do it from scratch now, or vice versa.

edjonesy, 2017-04-12:
I created a new micro sd card from scratch with 1.8.0 and it works perfectly with my set up, must've been a glitch when using the internal update option on the pre-release channel.

Thanks for the pointers and help.

Not a bug, just a glitch. Phew.

closed 2017-04-12 by edjonesy