RasPlex/RasPlex #576: Rasplex keeps freezing or losing Internet connection

FlyingPersian, 2017-04-22:
There's noting I really do. Sometimes it freezes while watching something, sometimes while in the main menu while I'm not actively using it. The Internet connection also sometimes doesn't come up after a reboot, sometime it goes away while idling. I'm using a rpi 2, with rasplex 1.8.0.148-573b6d73. I'm using a wired connection. Under network, it says active, but the connections window doesn't load up. Rebooting using my TVs remote sometimes takes a minute or two after I hit reboot (it's stuck in the menu then until it reboots). 
I'm using hyperion to with LEDs. 

PMS runs on a feenas server in the same network, also wired. It's at version 1.5.5.

g1t-dlanor, 2017-04-24:
Since you are having various freezing situations, even without any direct activity,  and since others do not report the same kind of issues, it seems unlikely that there is any specific program bug involved. Instead I suspect some more generic problem with your hardware. And general instability such as you describe is most often caused by either power supply problems or SD card corruption.

For the power supply the most common problems are insufficient current capacity and/or bad connectors/cables with too high resistance. An RPi2 needs at least 1.5 Ampere capacity with good cables for reliable operation. Personally I strongly recommend using a PSU with 2A capacity, and if you are powering additional hardware from RPi2 outputs you have to take that into consideration as well.

For the SD card corruption some known causes are power supply issues (as already described), low-class SD cards (good ones should be 'Class10' or 'UHS-1'), or incorrect shutdown procedures (like pulling power without first shutting down from menu).  The latter is a common risk if powering the RPi from USB connectors of a TV set, since this may cut power from such USB ports whenever the TV set is turned off.

IskanderNovena, 2017-11-08:
Old thread, but I have experienced freezes of raspberry when not actively using it. I have to unplug the power and connect it again to get it working.
No occurences during usage. Issue has happened two times in about three weeks.