RasPlex/RasPlex #588: Using RasPlex in offline mode ONLY, zero online connectivity

Superman185, 2017-08-02:
[Tell us about your issue, steps to reproduce, expected and actual behavior]
[Always attempt to reproduce your issue in the default skin]
[Enable debug logging in preferences and insert a link to your log-file]

**RPi model**: 2, model B v1.1
**RasPlex**: 1.7.1 & 1.8.0
**PMS**: Latest ver, Win10 
**Network**: ethernet & wifi
**Skin**: All

I am wondering how to setup RasPlex for offline use. I live in an area that has shotty reliability in extreme weather. My internet goes out several times a month and sometimes for days.  

I'm trying to get away from chromecast because it requires a 100% full time use of a internet connection.

I have been trying to configure this RasPlex for offline use and simulating an outage of my internet. This still allows me to connect to my local Plex server, however if I restart the PI and simulate a power loss (frequent as well) I can NO longer connect to my local media.  I can use any other device this way PCs, idevices, androids and still use Plex when my internet is down and a power loss reset.  I want that for RasPlex so this device can perform equally, like the Plex app does on those devices. I had to give up chromecast because it required a constant internet connection.  I thought RasPlex would be the solution.

I have put in the manual media server under >>> Preferences>Media Server>Manually specify server <<< to my local ip address and correct port.
I have made sure that my server firewall accepts all connections from the PI's IP address. I even disabled to try this out.
I have set my plex server for unsecure connections and set the IP for the PI in the advanced menu to "List of IP addresses and networks that are allowed without auth" (saved and rebooted) and still no luck. 

It should still be network discoverable.  Because it still has my login user name (all skins) when it's not connected to the internet and rebooted. Why is that?  So if it does have my login information, it still must have local IP/secure connection information of my network.  And so it's RasPlex requiring a internet connection.  Plain and simple with no obvious solution.

So why does RasPlex require a internet connection? Whether or not it's for the clock, or is it just to enable RasPlex? Like chromecast does.

Additionally I've done two guides that say "offline use" (one was for a car setup) and both require a internet connection in some form to access my Plex home server, and again over the local network. I've searched and searched and all I could get for "offline use" was to change the skin and browse the local media aka the flash drive, that's the car guide, with the "Black" skin.  There is nothing that I have found that makes RasPlex work without checking online to be enabled and, only looking for a local area connection.

I want it to work with no internet connection and offline from the internet.
There must be some way to accomplish this.  
So is there a way to shut RasPlex's internet requirement off? I don't care about the clock.  Plex does not require the correct clock setting on any device, or connection.
Where is programming/options file for the startup so it can be turned off?
Is there a add on that can do it?
Can any one please tell me how to set that up. Or suggest another app that works just as well as RasPlex that can accomplish that goal please.

Thank you.

Superman185, 2017-08-03:
Is there a offline only version of rasplex?  If not could it be done?

benjaminwolf, 2017-08-03:
Plex is authenticated at the corporate Plex.tv servers, so the authentication system is not designed to be operated in an offline mode. However, you can run Rasplex offline with a little bit of trickery on client and server.  But the initial login to the client and server needs to happen with an available internet connection so that the authentication tokens can be obtained from Plex.tv.

So while you have an internet connection to both server and Rasplex client, login to Rasplex, then turn on "auto login" in the Rasplex preferences.  http://forums.plex.tv/discussion/comment/1343974/#Comment_1343974

That should allow for login and playing of your local server content when your internet is out. Though, you may want to decide if a standalone system such as LibreElec is better suited to your purposes.

jimmypooh, 2020-06-09:
I'd like to add to the need for this as well.  Plex servers have the setting "List of IP addresses and networks that are allowed without auth"  This effectively could allow access to the server without authorization at all.  It seems the RasPlex client doesn't have the ability to use this feature without an active conneciton.  Can this be added?

I can confirm this is possible offline with clients such as an old android phone that does not have a cell or internet connection of any sort, and also an old Roku without an internet connection, so this is not specific to the Plex architecture.